dji-sdk/Onboard-SDK #137: M600 Pro with OSDK 3.3 getting "firmware is not supported" error

mattogodoy, 2017-06-28:
Hi!

I have a **DJI M600 Pro** updated to the last version (**1.0.1.20**) and I'd like to use the version 3.3 of the OSDK in a **Raspberry Pi 3**.

According to the release notes for 3.3 (https://developer.dji.com/onboard-sdk/documentation/appendix/releaseNotes.html) the Matrice 600 Pro should be compatible with this new version, but according to the Versioning Map (https://developer.dji.com/onboard-sdk/documentation/appendix/versioning.html) it's "_Comming soon_".

The problem is that when I follow all of the setup steps to run a sample (https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html) I'm getting a "**Your firmware is not supported by this version of the OSDK.**" error:

```
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.

ERROR/1 @ validateSerialDevice, L109: Succeeded to read from serial device

STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L525: Hardware =

STATUS/1 @ parseDroneVersionInfo, L526: Firmware = 3.2.15.62

STATUS/1 @ parseDroneVersionInfo, L529: Version CRC = 0x0

ERROR/1 @ initVersion, L317: Your firmware is not supported by this version of the OSDK.

ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
Vehicle not initialized, exiting.
```

I have been able to successfully run the sample apps for version 3.2, which confirms that the parameters are correctly set on DJI Assistant 2, and that the hardware connections between the flight controller and the Raspberry are correctly wired.

My question is: Is the Matrice 600 Pro compatible with the 3.3 version of the OSDK or not?

If it is, why am I getting this error?
If it's not, when will it be? (and please clarify this in the release notes!: https://developer.dji.com/onboard-sdk/documentation/appendix/releaseNotes.html)

Thanks a lot!

rohitsantdji, 2017-06-28:
Hi Matias,

This topic has been covered in #134 . Release notes have recently been updated with a link to the [version compatibility](https://developer.dji.com/onboard-sdk/documentation/appendix/versioning.html) page, and the frontpage of this github repo has also been updated with a copy of the version compatiblity matrix. To summarize the discussion in #134 - M600 Pro firmware supporting 3.3 should be out very soon, sorry for the delay!

closed 2017-06-28 by rohitsantdji

mattogodoy, 2017-06-29:
I have to start working next week in a project based on the OSDK, and I have to make a decision about waiting until 3.3 is released, or to develop based on 3.2.

What do you advise? Can you give us a time estimate?

rohitsantdji, 2017-06-29:
I checked with our firmware teams, and it seems the current planned date for M600/Pro firmware release is end-July. I'll update the release notes and the versioning matrix with this info soon!

mattogodoy, 2017-06-29:
Ok! That's good to know.
Thanks a lot!