dji-sdk/Onboard-SDK #138: RTK Fix status not available in 3.2

mattogodoy, 2017-06-29:
Hello!

I'm working on an app based on OSDK 3.2 with a **Matrice 600 Pro** with a D-RTK module, and I'd like to get all of the RTK information, specially the **Fix** status. But as I can see in [DJI_Type.h](https://github.com/dji-sdk/Onboard-SDK/blob/3.2/osdk-core/inc/DJI_Type.h), the RTK data struct does not contain any status attribute:

```
typedef struct RTKData
{
  uint32_t date;
  uint32_t time;
  float64_t longitude;
  float64_t latitude;
  //! @warning the 'Hmsl' field will be renamed in a future release.
  float32_t Hmsl;

  float32_t velocityNorth;
  float32_t velocityEast;
  //! @warning the 'velocityGround' field will be renamed to velocityDown in the next release.
  float32_t velocityGround;
  
  int16_t yaw;
  uint8_t posFlag;
  uint8_t yawFlag;

} RTKData;
```

Is there a way of getting the fix status on the 3.2 version of the OSDK?
I can see that it is possible using version 3.3, but it's still not available for the 600 Pro. I know that it will be released soon, but I'd love to have more information about this. Soon could mean one week or two months. Can we get a time estimate?

Thanks!

rohitsantdji, 2017-06-29:
The RTK fix was introduced in the new firmware, so it is not possible to get that data with OSDK 3.2.

mattogodoy, 2017-06-29:
Good to know.
Thanks!

closed 2017-06-29 by mattogodoy