dji-sdk/Onboard-SDK #139: Waypoint Event Callbacks

hdino, 2017-06-29:
Does the OSDK v3.3 generate callbacks if a waypoint is reached during a waypoint mission? I tried this with OSDK v3.2, but it did not work. In the current version there is still a todo note in the code: [dji_vehicple.cpp:1267](https://github.com/dji-sdk/Onboard-SDK/blob/3.3/osdk-core/api/src/dji_vehicle.cpp#L1267)

rohitsantdji, 2017-06-29:
Yes, it will call the wayPointEventCallback you register with missionManager->wpMission at the start of the program. This callback is called on all waypoint events, including waypoint reached event - see the WayPointIncidentType enum [here](http://developer.dji.com/onboard-api-reference/namespaceDJI_1_1OSDK.html#enum-members).

rohitsantdji, 2017-07-05:
Closed due to lack of activity, if there are further questions please re-open the issue. Thanks!

closed 2017-07-05 by rohitsantdji