dji-sdk/Onboard-SDK #140: A3 low-battery actions

hdino, 2017-06-29:
How does the A3 flight controller behave if the battery level becomes low while the aircraft is controlled via the OSDK? In the Mobile SDK documentation I found something about battery level thresholds for return-to-home and land-in-place actions. Does the flight controller take away control from the ODSK at some point? Can the user alter these thresholds?

rohitsantdji, 2017-06-29:
The behavior is the same for OSDK as well, at the low battery level and the critical battery level the flight controller will override the SDK and execute some pre-configured actions. These actions (and the battery levels that count as "low" and "critical") can be configured in DJI Assistant 2's Flight Settings -> Battery page.

rohitsantdji, 2017-07-05:
Closed due to lack of activity, if there are further questions please re-open the issue. Thanks!

closed 2017-07-05 by rohitsantdji