dji-sdk/Onboard-SDK #142: OSDK v3.3 throwing up error on running sample

taurean1192, 2017-07-03:
I am getting the following error upon running the sample for telemetry. Earlier my flightcontrol sample was running fine but now even that is throwing error.

[image]

Some help will be really great.

rohitsantdji, 2017-07-05:
@taurean1192 it appears that there is no data on the line. Could you provide some more information to help us debug:

1. Is this a Pi2 or a Pi3 or a Pi zero?
2. Do you have permissions to write to the serial port (did you run sudo usermod -a -G dialout $USER ?)
4. After upgrading firmware on your A3/N3, did you ensure that the baud rate to the correct value, and reboot the flight controller after clicking enable API control?
5. Could you try the sample at baud rate 230400?

A similar issue is also being tracked here : dji-sdk/Onboard-SDK-ROS#98

taurean1192, 2017-07-11:
@rohitsantdji  
1. This is a Pi3 running Raspbian
2. Yes, I do have permissions to write to the serial port.
3. I am connecting to an A3
4. Yes, I had checked for the API control. I had cross-checked for matching baud rates as well, the flight control sample did run as a matter of fact
5. I tried changing it to 230400 and check, the output was same as last time.

adamwathieu, 2017-07-11:
I have a similar issue, any help would be great:
https://github.com/dji-sdk/Onboard-SDK/issues/143

chutsu, 2017-07-12:
I was having the same problem but I think I found a solution, the approach I took to fix this issue is to **simply plug the USB-UART directly into the computer's USB port** (not USB hub).

taurean1192, 2017-07-14:
@chutsu I did that.. still the same issue.. as 
[image]

chutsu, 2017-07-14:
When you copied the `UserConfig.txt` did you actually put in your own APP ID and APP KEY? in your above screenshots it does not look like you edited it.

taurean1192, 2017-07-15:
If you know in 3.3 it doesn't show up when the code executes. Yes, I have
my API ID n Key

closed 2019-05-28 by hummelslei