dji-sdk/Onboard-SDK #143: Problems with connecting Raspberry Pi successfully to M100 through serial port

adamwathieu, 2017-07-06:
Currently I am trying to mount a raspberry pi 3 as my onboard device onto my M100. I am connecting my pi by connecting the UART_CAN2 port to the GND, TXD, and RXD pins of my pi with the provided UART serial cable. I am using the 3.1 branch of the SDK provided by DJI on Github, which is compatible with the M100. When I try to execute djiosdk-linux-sample from my build directory, this comes up:

```
User Configuration read successfully. 

These are your User_Config settings.
Serial port = /dev/ttyS0
Baudrate = 115200
Drone/FC Version: M100
SDK Version: 3.1
App ID: MY_APP_ID_XXXXXXX
App Key: MY_APP_KEY_XXXXXXXXXXXXXXXXXXXXXXX

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make changes.

STATUS init,line 51: Attempting to open device /dev/ttyS0 with baudrate 115200...
STATUS init,line 60: ...Serial started successfully.

Attempting activation..
Automatic activation successful.
Failed to obtain control.
Your RC mode switch is not in mode F. (Is the RC connected and paired?)
This program will exit now.

```
I am aware that this question has been asked before, but I haven't found any compelling answers. My remote is on mode F, and it is recognized as so on the Assistant application. However, I know that is not the problem. I have heard from other people online (https://github.com/dji-sdk/Onboard-SDK/issues/77), that the remote controller and its mode is not the core of the problem. Does anyone know how to resolve this? Any help would be great.

Thanks, AWATHIEU

jhmoon999, 2017-07-11:
I am in the same situation except I am using the 3.2 branch of the Onboard SDK for the Matrice 100. When I execute the djiosdk-linux-sample, I get the same error, except with an additional:

ERROR parseDroneVersionInfo, line246: Drone version was not obtained. Please restart the program or call getDroneVersion

which seemed to be a common error ([#88](https://github.com/dji-sdk/Onboard-SDK/issues/88)), but wasn't really resolved. 

It's possible my UserConfig.txt is incorrect. Could anyone share their format?

adamwathieu, 2017-07-11:
This is my UserConfig.txt:

```
ID: XXXXXXXX
KEY: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
DeviceName: /dev/ttyS0
BaudRate: 115200
Version: versionM100_31
```
Not sure if this is the issue, but I have 3.1

drieshulens, 2017-07-12:
I also could not connect via Serial port = /dev/ttyS0, so I used a USB to serial converter and it worked without a problem. Also, if you are able to make a connection, make your baudrate as high as possible otherwise you will have problems receiving all messages correct (in my case the RC messages where not send correctly when my baudrate was to low)

Hope this helps !

adamwathieu, 2017-07-12:
@drieshulens,
Thank you!
I did try to use a USB to serial converter and am now using the /dev/ttyUSB0 port. I changed the baud rate to 921600, but I still have the same issue. Do you have a Pi 3 and version 3.1 of the SDK?Any suggestions?

drieshulens, 2017-07-12:
Im using version 3.3 (but you should use 3.2 for the M100) and 230400 as baud rate. The highest baud rate gave me problems connecting with the raspberry. Is API control marked via the DJI Assistant ?

adamwathieu, 2017-07-12:
The Enable API Control checkbox is checked, and our baudrate is now 230400. Still nothing. Our SDK is set to 3.2 now.

drieshulens, 2017-07-12:
IS the M100 connected with the flight simulator in DJI assistant ? And is your controller in F mode ?

adamwathieu, 2017-07-12:
We were not aware that the simulator had to be on, but when we turned it on, the same error came up. Also, our remote has always been on F.

drieshulens, 2017-07-12:
No it doesn't but to test things it is easier in simulation first. Maybe a stupid question but is your remote also powered on ? And can you fly in simulation ?

rohitsantdji, 2017-07-12:
Please post pictures of your connection, also as @drieshulens suggested could you try 3.2 since that's the most stable M100 branch?

adamwathieu, 2017-07-12:
The drone and the controller are powered on, and Assistant tells me that it is in mode F. It is also on mode F on the controller. Could switching to a Pi2 help? Maybe the SDK is unstable for the Pi3 because it is newly released?

rohitsantdji, 2017-07-12:
The 3.2 branch is stable for M100, and we've seen some members of the community have success with using it on a Pi 3. Switching to a Pi2 will not help since the pi2 has a version of gcc that doesn't support c++11 out of the box. Please post pictures of your connection to help us debug this further, since this seems like a connection issue.  (The activation successful message is not accurate on non x86_64 systems for OSDK 3.1/3.2)

adamwathieu, 2017-07-12:
I cannot provide pictures right now, but we redid everything under the raspberry-pi support branch of the onboard sdk repo, and we still had the same outcome.

jhmoon999, 2017-07-12:
I've had the same setup as Adam and have double-checked the things suggested by drieshulens. I get alternating errors whenever I run the sample: 

[image]

I have not tried using a USB-TTL cable, but I will purchase one.

jhmoon999, 2017-07-12:
This is my DJI Assistant 2 settings for SDK:

[image]

jhmoon999, 2017-07-12:
@adamwathieu I originally had your error, but then I realized my RX and TX pins of my Pi were connected to the RX and TX pins of the drone. I swapped them to correct it and got this new errors above. Are your pins swapped by any chance?

adamwathieu, 2017-07-13:
Yes that was my issue! Intuitively I thought the RXD would go to the RXD and the TXD would go to the TXD, but after switching them over, it worked! Thank you @jhmoon999!!!

adamwathieu, 2017-07-13:
Ok,
Here are the pictures:as

closed 2017-07-13 by adamwathieu

jhmoon999, 2017-07-13:
@adamwathieu Glad I could help! 
@rohitsantdji If Adam's worked just like that, I think I'm very close to getting it to work too. All I need is to fix this "ERROR parseDroneVersionInfo, line 246: Drone version was not obtained. Please restart the program or call getDroneVersion" part. Do you have any suggestions?

adamwathieu, 2017-07-13:
@jhmoon999, try this:
osdk-wrapper/src/LinuxSetup.cpp, uncomment the call to api->setVersion() on line 201.

jhmoon999, 2017-07-13:
I remember doing that with the Raspberry Pi 2 and the raspberrypi-support branch. I'm currently using the Raspberry Pi 3 and the 3.2 branch, and the osdk-wrapper/src/LinuxSetup.cpp is different; there is no api->setVersion() on line 201.

adamwathieu, 2017-07-13:
Yeah so I had that too, but then redownloaded 3.3 and then checked out the raspberrypi-support branch. Then I did the above command.

jhmoon999, 2017-07-13:
So what you're saying is, implement the raspberrypi-support branch on the Raspberry Pi 3. And it worked? I'll give it a try. Thanks.

adamwathieu, 2017-07-13:
Yes, that's what we're doing at least.

jhmoon999, 2017-07-13:
@adamwathieu Thanks! I did as you suggested and progressed a little further.

Out of the 20 times I run ./djiosdk-linux-sample, 18 of them are either "Your RC mode switch is not in mode F" error, "There was an error with the activation command. This can happen due to a variety of reasons" error, or "Make sure DJI GO is turned on and connected to the internet" error. 

But the other 2 times, a connection is made but with "Error in setControl API function." 
[image]

Has anyone encountered this error?

Martintl, 2017-07-20:
@jhmoon999 
Hello , i have the same problem with M100 . "activation didn't go through" and "Error in setControl API" . Can you please write what should i do ( step by step , i'm very new in this theme ) . Thank you !

jhmoon999, 2017-07-20:
@Martintl I wrote some tips here: #148 

Try them out first. My main problem was that my iOS phone could not do activation whereas an Android phone can.

yekaifeng, 2017-10-06:
Don't know if it is helpful, the GPIO-connected UART device is broken on Pi 3.  The mini-uart doesn't have a separate clock divisor that might result in arbitrary modification of the effective baud rate. Here are the reference from official rsp site:

- https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=137932
- https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=138223