dji-sdk/Onboard-SDK #145: Cannot activate M100's using Onboard-SDK 3.2

brentalanmiller, 2017-07-10:
We've been unable to activate some new M100's that we're using. We first noticed this last Friday and thought that maybe the activation server was down, but we're still having issues today. Even if we checkout the latest from the 3.2 branch (b717cd6) and run the sample application, it still fails.

```
$ ./djiosdk-linux-sample
User Configuration read successfully. 

These are your User_Config settings.
Serial port = /dev/ttyUSB0
Baudrate = 230400
App ID: XXXXXXX
App Key: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make changes.

STATUS init,line 56: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS init,line 65: ...Serial started successfully.
STATUS setFrequencyCallback,line 1199: Frequency set successfully
STATUS validateSerialDevice,line 118: Succeeded to read from serial device
STATUS parseDroneVersionInfo,line 430: Device Serial No. = 041DE21495
STATUS parseDroneVersionInfo,line 433: Hardware = M100
STATUS parseDroneVersionInfo,line 435: Firmware = 3.1.10.0
STATUS parseDroneVersionInfo,line 439: Version CRC = 0xA6453AAC

Attempting activation..
Your activation did not go through. 
This is a new device. 
Make sure DJI GO is turned on and connected to the internet 
so you can contact the server for activation.
This program will exit now. 
```
This seems like it was working fine a couple of weeks ago, did something change to the authentication server when you released 3.3?

We're using the latest DJI Go app for iOS, 3.1.13, and we have a good connection to the aircraft and the iPhone has a good connection to the internet.

Can anyone help out with this?

adamwathieu, 2017-07-11:
I have a very similar issue. Any help would be great:
https://github.com/dji-sdk/Onboard-SDK/issues/143

brentalanmiller, 2017-07-11:
We were able to work around the issue by using the Android version of the the DJI Go app when we activated the Onboard SDK -- the problem seems to be isolated to the current iOS version of the DJI Go app (3.1.13).

brentalanmiller, 2017-09-21:
No longer an issues with recent versions of DJI Go. Closing issue.

closed 2017-09-21 by brentalanmiller