dji-sdk/Onboard-SDK #146: Buffer overflow happened on RPi3 connecting to N3 board (OSDK 3.3)

kevinphys, 2017-07-11:
I have tested on Ubuntu 16.04 within virtualbox which could successfully connect to N3 board before. While simply moving USB-TTL port to Raspberry Pi 3 with Ubuntu MATE 16.04, the error occurred:

```
$ ./djiosdk-telemetry-sample
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.

ERROR/1 @ validateSerialDevice, L109:  Succeeded to read from serial device

STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L525: Hardware =

STATUS/1 @ parseDroneVersionInfo, L526 Firmware = 3.2.36.8

STATUS/1 @ parseDroneVersionInfo, L529 Version CRC = 0x0
*** buffer overflow detected ***: ./djiosdk-telemetry-sample terminated
Aborted
```

- Enable API checked
- baudrate cross-checked (with 230400)
- Port permission checked

All, including firmware and OSDK, in the latest version.

Could any issue happened with USB port on Raspberry Pi? What happened to the connection between N3 and RPi3?

adamwathieu, 2017-07-12:
Hey!
Could you please take a look at my issue? It seems like you have gotten a little further than I have:
https://github.com/dji-sdk/Onboard-SDK/issues/143
Thanks!

rohitsantdji, 2017-07-12:
@kevinphys This seems like a RasPi-related issue for 3.3, sounds like it is similar in concept to #135 . Is it possible for you try with a different computer? 3.3 raspberry pi support is on the roadmap, but it won't be out immediately.

kevinphys, 2017-07-14:
@adamwathieu I saw that what you are using is raspberry-pi branch and 3.2, which means you are on the top of Raspbian. Have you ever tried 3.3 with Ubuntu 16.04 MATE on RPi3?

@rohitsantdji Thanks! So which means the last one we can try is STM32 instead of computer. I followed the issue on #135, unaligned memory access looks critical for developing onboard-SDK with RPi on DJI.

I am also considering Raspbian brand but is the version only up to 3.2? It seems like only 3.3 supports MFIO.

rohitsantdji, 2017-07-20:
Yes, only 3.3 supports MFIO.

hummelslei, 2019-05-28:
Close this issue now. If you still have more issues related to DJI SDK, please help to fill in this form: 
https://formcrafts.com/a/dji-developer-feedback-en to report them to DJI Developer Support. Thanks for your understanding and cooperation.

closed 2019-05-28 by hummelslei

afwilkin, 2020-01-05:
@kevinphys 

Were you able to connect the N3 straight to the raspberry pi? Or did you have to use an external power supply? We are not able to communicate with the N3, and we are suspecting it is because the raspberry pi does not have enough power output on the usb port