dji-sdk/Onboard-SDK #147: Unusual communication problem with A3

Korhanyuz, 2017-07-12:
Hi,
We have tried and succeeded to use an USB-UART (TTL) converter for connecting a PC to A3’s (matrice 600) API port. We implemented the open protocol and was able to send commands like Activate, Auto Take Off, Return To Home, Design a waypoint mssion etc. successfully. For that time, we were quite happy with the communication stability as, I can say, not even one byte was missing. Communication line was solid and stable…

After some time, we tried it again and there was no response from drone API port. The difference was not the telegram we were sending. Because we were using exactly the same software we had personally built for this. It cannot be different at all.

That’s why, we were suspicious about the communication and interference problem.

Then we found something weird: if we touch the Tx/Rx cable by hand (the cable’s plastic, not the live copper), the communication line shows us that we have incoming bytes, mostly zero, and a few other bytes occasionally. This is a clear evidence that the problem was the communication line.

It is still an open question; how it was working absolutely perfectly before (a few months ago) and not working at all today???

But anyhow, what is your suggestions to maintain a stable healthy communication?

We use USB – Uart TTL converter.

PS. We have also tried to take power off of our laptops (to isolate them from mains power) etc., makes no difference.

Just to note: Once I saw that: we received a byte “FF” when air conditioner relay clicked. So this is an interference problem which mostly bad effecting if comm line is prone to interference. We need to make our communication more immune to environmental interference.

There was a software update on drone side but I guess it cannot have a bad effect for this, as this seems to be mostly a hardware issue.

Your comments please…

Thanks

rohitsantdji, 2017-07-12:
This does seem to be a hardware issue more than anything else - I believe you have diagnosed it correctly as interference from external sources. The other things I would recommend checking are:

1. Is API control enabled, and the baudrate correct? These selections will reset after a firmware update, so need to be manually set up again.
2. Can you try with a different cable, something that has a Prolific PL2303A chip instead of an FTDI (in case your current USB-serial cable has an FTDI)?

chutsu, 2017-07-12:
Out of curiosity

1. Is your USB-UART directly plugged into the PC's USB port or are you using a USB Hub?
2. Are you using USB 3.0 Cameras by any chance?

Korhanyuz, 2017-07-14:
@rohitsantdji 
We have added 2 bluetooth to UART converter, 1 to pc and 1 to Drone side and it works perfectly fine.
I think the decoupling devices wirelessly did the trick for us.

@chutsu 
1. Yes. No hub
2. No

Thanks

closed 2017-07-20 by rohitsantdji