dji-sdk/Onboard-SDK #149: Vertical velocity control bugs

bluesll, 2017-07-19:
I use SetMovementControl() to control the vertical velocity of the UAV, but when I set a negative vertical velocity( to make the UAV approach the ground), the UAV just fall down and stop working immediately, even though the magnitude is tiny.
I wonder if there are some bugs in Onboard-SDK or maybe in the firmware.

rohitsantdji, 2017-07-19:
Please provide more info:

1. What onboard computer are you using? 
2. What DJI product are you connecting to?
3. What is the baud rate?
4. What version of the SDK are you using?
5. What are the magnitudes of the commands you are sending? The entire sequence would be helpful.

bluesll, 2017-07-22:
1. onboard computer: 
Although I don't think it really matter, I was using Intel NUC, and the same problem also occurred for DJI manifold and jetson-TK1.
2. DJI product
M100 with latest firmware.
3. baud rate
230400
4. SDK version
3.2, and I have asked some people using 3.3 who have found the same problem.
5. Magnitudes and sequence
I was sending command by setMovementContol with control mode flag 0x41 (HORI_VEL, VERT_VEL, YAW_RATE), when I set vertical velocity to a negative value, no matter what magnitude it is, my aircraft just fall down and stop working immediately.
The sequence of my operation is that I send takeoff command to m100 and when it is hovering stably, I send a negative vertical velocity command, then problem just occurred.
BTW, I don't have GPS signal but use guidance instead and guidance was functioning properly.
Many people have found this problem, and some people said it is a bug in the m100 firmware.

rohitsantdji, 2017-08-14:
Sorry for the late reply! Yes, that is indeed an M100 firmware bug. We've had some success with a workaround, here is a possible solution: Instead of (takeoff ->hover->negative vertical velocity) sequence of commands, try (arm -> positive vertical velocity greater than 0.5 m/s -> hover -> negative vertical velocity). Essentially, the drone should receive a positive vertical  velocity of more than +0.5 m/s during a flight for the negative vertical velocity to work correctly. Could you try this out?

closed 2019-05-28 by hummelslei