dji-sdk/Onboard-SDK #152: Waypoint mission fails if LB2 is off or signal is lost

flytrex-vadim, 2017-07-20:
With LB2 remote turned off or out of range, the system will obtain control, complete automatic takeoff to the height of one meter, but then refuses to execute waypoint mission (receives the ACK with the "no error" code but keeps hovering at 1 meter)

We tried setting "Enable SDK Failsafe Action" together with "SDK Failsafe Action: Hover" as described in comment to issue #118 . 

When the LB2 signal is lost during the execution of waypoint mission, the state changes to "Hovering" and it stops in place. 

How can we make it start/continue the waypoint mission regardless of the LB2 state ? 

Seeing this both in simulation and in real flight
[SDK v3.2, A3 FW v1.7.0.5]

rohitsantdji, 2017-07-20:
When you set up the waypoint mission with a call to `waypoint->init(WayPointInitData , ...)` did you set `RCLostAction = 1` within the WayPointInitData struct? (See 3.3 docs [here](http://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1WayPointInitSettings.html#aa46da1cfda3af725541e1b897c8a998f), the WaypointInitSettings struct in 3.3 corresponds to WayPointInitData in 3.2). In the 3.2 examples, it is set to 0 by default which means when the RC is lost the mission will exit.

flytrex-vadim, 2017-07-21:
Thanks, that solved it!

closed 2017-07-21 by flytrex-vadim