dji-sdk/Onboard-SDK #153: M600 uart connection with Raspberry Pi 3

rocklinsuv, 2017-07-21:
Hi there,

I'm currently doing some preliminary research on M600 with RP3 onboard computer. I've found a website describing the hardware configuration: [link](https://developer.dji.com/onboard-sdk/documentation/hardware-setup/index.html). May anyone please advice me which connection I should choose (I have a feeling that the A3/N3-Manifold might be the one I'm after)? Is it a direct pin-to-pin connection or should I get a TTL-USB cable too? Many thanks.

Cheers,

rohitsantdji, 2017-07-21:
Either one should be fine for an RP3. You should also take a look at @jhmoon999's excellent [set of tips](https://github.com/dji-sdk/Onboard-SDK/wiki/Tips-on-connecting-the-Raspberry-Pi-to-the-Matrice-100) for connecting raspberry Pi's with DJI products (the guide mentions connecting to an M100 but it is valid for the M600 as well).

rocklinsuv, 2017-07-21:
Thanks for your quick reply bro. I will try a direct connection (RP3's TX to M600's RX and vice versa) then. And the tips are really helpful too :smile:

closed 2017-07-21 by rocklinsuv