dji-sdk/Onboard-SDK #155: onboard SDK backwards compatibility

greg-draganfly, 2017-07-24:
I am currently developing an application using the onboard SDK3.2.  When 3.3 becomes available for my platform will it continue to support onboard SDK 3.2?  Suppose I am using an M600 and update the firmware to support 3.3 does this action mean I need to rewrite my application to use onboard sdk 3.3 or can I continue to use 3.2?

rohitsantdji, 2017-08-09:
Hi Greg - you will need to update your application to use OSDK 3.3 when you upgrade to a compatible firmware, because the firmware changes a number of protocols and older OSDK versions will behave erratically with new firmware.

closed 2017-08-09 by rohitsantdji

rohitsantdji, 2017-08-16:
Hi @greg-draganfly, FYI - M600 firmware has been updated to support OSDK 3.3.1!

greg-draganfly, 2017-08-16:
I see in the release notes "This release introduces backward compatibility for the DJI M100".  Only the M100?  Why not the M600 as well?

Is the update for the M600 mandated at this point, or can I choose to defer it for a while?

rohitsantdji, 2017-08-16:
The M600 was waiting on a firmware update at the time of the OSDK 3.3.1 release, so the release notes do not mention compatibility. Yesterday, the necessary firmware update was made available - we'll be updating the docs very soon to reflect this. 

The firmware update won't be mandatory, so you can continue to use OSDK 3.2 with the old firmware. For any new development with the OSDK we highly recommend switching to OSDK 3.3 when possible. When you do decide to switch, you will need to upgrade the firmware on the M600.