dji-sdk/Onboard-SDK #156: Transparent Transmission + Raspberry Pi 3 + Sensors

mtomoto, 2017-07-25:
Hello everyone,

I am using a Matrice 600 with some sensors commanded by a Raspberry pi. I am trying to control my sensors using my Mobile device, through Transparent Transmission (Mobile -> Raspberry -> Sensors). Have someone ever tried this before? I am with some issues to make it work and I don't know even if I am changing the right files. Does someone have a tutorial, project to share or even some tips? 

I am using SDK 3.2 and Rpi3 with Raspbian.

In a next step I want to also receive the data from the sensors and see in "real time" on the screen of the cellphone.

rohitsantdji, 2017-07-25:
Maybe you can start by following the discussion in #112? The general concept of your usage scenario has been tested in-house; you should be able to set it up exactly as the [docs](http://developer.dji.com/onboard-sdk/documentation/M100-Docs/application-development-guides/data-transparent-transmission.html) suggest. The samples in 3.2 (osdk-wrapper/LinuxMobile.cpp) use a simple protocol to trigger OSDK commands from the MSDK; you can follow this in DJI_API.h and in the LinuxMobile wrapper.

mtomoto, 2017-07-25:
@rohitsantdji Thank you very much for your answer. I saw this post before asking. I was able to run the sample both from TT and Onboard as it is said in DJI docs. However, I don't know if the newer one that I am creating is right. Do you know how is the proceed to create a new "sample", even in the Onboard Sdk? I want to make sure I am going in the right way, which I don't know exactly.

closed 2019-05-28 by hummelslei