dji-sdk/Onboard-SDK #157: waypoint event callback function got invalid data buffer pointer

magenious, 2017-07-27:
I used setWaypointEventCallback() to received waypoint mission event as follows:
pVehicle->missionManager->wpMission->setWaypointEventCallback(
       WaypointEventCallback, NULL);
and my WaypointEventCallback() defined as follows:
[image]
then I got problem when drone arrived the first waypoint, yes it received event and the callback function was called, as the above picture illustrated: the data length was 24, that is correct(recvFrame.recvInfo.len=24). But the data buffer pointer was invalid, that was,  recvFrame.recvInfo.buf was invalid.

Hope you confirm this and fix it, thanks a lot.

magenious, 2017-07-27:
BTW, I used osdk of V3.3.0 in windows, the firmware of A3 was version 3.2.36.8

magenious, 2017-08-02:
Did anyone notice this issue? Please comment if you confirmed, thank you.

rohitsantdji, 2017-08-02:
What serial drivers are you using to run this on Windows?

magenious, 2017-08-03:
@rohitsantdji Thanks for your comment. I used the following driver:
[image]

I'm confused that I could received all data correctly form Telemetry or broadcast, but received data point was invalid in the waypoint event callback function.  Greatly appreciate if you could look through. Thank you.

magenious, 2017-08-04:
Hi, any updates? @rohitsantdji

rohitsantdji, 2017-08-09:
@magenious - I looked at your screenshot and I think there is a slight misunderstanding of how `recvFrame` works.

1. `recvFrame.recvInfo` stores just metadata about the received packet. In most cases at the application level, the only relevant parts of the `recvFrame` are `cmd_set`, `cmd_id`, `len`. The `buf` attribute is used for storing the **send** buffer (not the received data) in a very specific scenario - the obtain/release control call where you need state information to figure out the meaning of the ACK.

2. `recvFrame.recvData` is what you really want to look at - the `raw_ack_array` attribute stores the received ACK during the Protocol/Link layer parsing and the application layer then casts that raw ACK into an ACK type that the current call expects. So in your case, you only need to access the `recvFrame.recvData.wayPointReachedData` attribute and parse it according to the [API reference](http://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1ACK_1_1WayPointReachedData.html) (you'll also need [this](http://developer.dji.com/onboard-api-reference/dji__mission__type_8hpp.html) reference).

This should enable you to solve your issue.

magenious, 2017-08-10:
@rohitsantdji Thanks for your detailed reply, now I realized the problem now, and my test also verified what you said, thanks a lot for your help. ^_^

closed 2017-08-10 by magenious