dji-sdk/Onboard-SDK #158: SDK Restart

hdino, 2017-07-28:
When using the OSDK 3.3 I delete the vehicle object on cleanup. However, when I restart my program it is unable to communicate with the A3 flight controller anymore. The following error messages show up:

```
STATUS/1 @ init, L37: Attempting to open device /dev/ttyS4 with baudrate 230400...
STATUS/1 @ init, L47: ...Serial started successfully.
ERROR/1 @ parseDroneVersionInfo, L368: Drone version was not obtained. Please restart the program or call getDroneVersion
ERROR/1 @ getDroneVersion, L913: Drone version not obtained! Please do not proceed. 
Check if your serial connection is okay, and if your activation status is okay.
ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
```

If I power off the copter and power it on again, the program can establish a new connection successfully. Is there a way to reset the controller in pre-authentication state (or something like that) and avoid the power cycle?

hdino, 2017-07-31:
Apparently, it was my fault... After cleaning up the code it's working :)

NielsPaul, 2017-08-23:
The issue is still existent in OSDK 3.3.1. It appears, when a connection has been successfully established and the program is closed afterwards. If you then want to reconnect to the OSDK, the above mentioned error messages are printed. Only a complete restart of the drone is solving this problem.
Is there maybe a safe way to close the connection to the sdk properly ?

rohitsantdji, 2017-08-23:
Hi @NielsPaul - we are not able to reproduce this issue. Could you provide more details regarding your setup? Some useful details would be drone/FC hardware, firmware version, onboard computer architecture, and whether you're running sample code or your own code.

NielsPaul, 2017-08-24:
Hey,
here are more information about our setup:
Drone: S100 with A3 Pro, Firmware 3.2.36.8
Onboard Computer: Intel Joule with Ubuntu 16.04, connected via GPIO UART (/dev/ttys4)
Code: Telemetry_Sample

When i check the serial connection, the flight controller is still sending data (see attached log) after the first run. However, only a restart enables me to reconnect (once) to the drone.

### 1.Run: Everything works fine!

& ./djiosdk-telemetry-sample ~/Onboard-SDK-3.3/UserConfig.txt
Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L36: Attempting to open device /dev/ttyS4 with baudrate 230400...
STATUS/1 @ init, L46: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L525: 
STATUS/1 @ parseDroneVersionInfo, L526: Firmware = 3.2.36.8
STATUS/1 @ parseDroneVersionInfo, L529: Version CRC = 0x0
STATUS/1 @ activate, L834: 
STATUS/1 @ activate, L847: Activation successful
| Available commands:                                            |
| [a] Subscribe to telemetry and print                           |
...
Done printing!

STATUS/1 @ removePackage, L423: 
STATUS/1 @ removePackage, L423: 
STATUS/1 @ removePackage, L423: Remove package 2 successful.
STATUS/1 @ removePackage, L423: 

### # 2.Run directly afterwards:

& ./djiosdk-telemetry-sample ~/Onboard-SDK-3.3/UserConfig.txt
Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L36: Attempting to open device /dev/ttyS4 with baudrate 230400...
ERROR/1 @ _serialOpen, L154: m_serial_fd = 3
STATUS/1 @ init, L46: ...Serial started successfully.
ERROR/1 @ parseDroneVersionInfo, L368: Drone version was not obtained. Please restart the program or call getDroneVersion
ERROR/1 @ getDroneVersion, L913: Drone version not obtained! Please do not proceed. 
Check if your serial connection is okay, and if your activation status is okay.
ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
Vehicle not initialized, exiting.

closed 2018-06-11 by hdino

adam3039, 2018-07-08:
Has anyone figured out a solution for this? I see that the ticket has been closed, but I'm getting the same issue when switching from running a build of my script, and then running afterwards in debug mode. I'm imagining its because the initial script is still connected to the flight controller somehow. Is it possible to disconnect somehow?

hdino, 2018-07-08:
I closed the ticket since the restart works for me with the current SDK. Basically, I adapted the Linux examples, i.e.
1) Create an instance of DJI::OSDK::Vehicle
2) Call its activate method
3) Subscribe to topics
4) Obtain control and fly around
5) Release control
6) Unsubscribe topics

adam3039, 2018-07-08:
Thanks, I think I was missing the bit where I released control. That'd do it ;)