dji-sdk/Onboard-SDK #160: TOPIC_HEIGHT_FUSION returns 0.0 every time

mk2017, 2017-08-02:
not sure what's wrong here, i'm able to get all other telemetry that I care about except this. Each time I request this data, it returns 0.0f

mk2017, 2017-08-02:
in simulation - this message seems to work - it only seems to misfunction when in real flight

zliDJI, 2017-08-03:
Hi, when you say real flight, do you mean take off and read the height and it gives 0?

mk2017, 2017-08-03:
yes, exactly.

mk2017, 2017-08-03:
fyi my workaround for now is to snapshot altitude when i take off and then subtract that value out to get my height (i need this value for vertical control)

mk2017, 2017-08-03:
Also, height is not explicitly listed here:
https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-telemetry.html

so maybe it's not a valid message to trust right now, but it's in the codebase so I wanted to use it if possible.

zliDJI, 2017-08-03:
Thanks for the info. We'll investigate this on our side.

magenious, 2017-08-09:
@mk2017 @zliDJI  Hi, all, I got this issue exactly a few days ago, and i could give more information:
1) the TOPIC_HEIGHT_HOMEPOOINT is correct in telemetry while TOPIC_HEIGHT_FUSION is always 0;
2) the height from broadcast data was also correct, now i just use this height to avoid the bug
3) Yes, in simulation the TOPIC_HEIGHT_FUSION is correct, but in real flight, it would be 0 always.

Hope this issue could be fixed soon then i could totally use telemetry in my code, thanks.

justwillim, 2017-08-09:
hi all,
The height fusion in telemetry means relative height fused with VO sensors and ultrasound sensors.
Since A3/N3/M600 do not have these sensors, in a real flight, this telemetry will return 0.
In M210/RTK, this telemetry will work.
In broadcast data, the height is equal to telemetry ALTITUDE_FUSED this is a GPS/barometer based data telemetry, almost works every time. 
Thanks

closed 2017-08-09 by rohitsantdji