dji-sdk/Onboard-SDK #162: Altitude values from the BroadcastData Structure not consistent when flying near a wall

krunal2103, 2017-08-10:
I am flying my quadcopter for an application which demands the quadcopter to fly near a wall.
I am using the altitude values from the PositionData Structure which is a part of BroadcastData structure as feedback to maintain my altitude. The aircraft maintains the altitude for the whole part when its approaching the wall. But the moment it is close to the wall (around 900 mm from the wall), it looses altitude. This change in altitude is not reflected in the altitude readings from BroadcastData Structure. Its because of this I am unable to hold my altitude. Is there any other function or values which I can read from SDK and that they might not do this?

My aircraft is using an A3 flight controller.

rohitsantdji, 2017-08-14:
Hi Krunal,

That behavior is likely due to the barometric pressure being a major component of the fusion that results in the altitude value; as you get closer to the wall, the airflow disruption is enough to cause the barometric reading to drop by a few feet. Does that match your observation? Also, do you have a strong GPS fix near the wall?

rohitsantdji, 2017-08-23:
Closing due to inactivity. Please re-open if you have follow-ups.

closed 2017-08-23 by rohitsantdji