dji-sdk/Onboard-SDK #163: Land command results in "hard landing" when done on a hill of over 20m

flytrex-vadim, 2017-08-13:
Hi,

A question regarding the land command. We observed that after initiating automatic landing,  the Matrice 600 descends with the velocity of 3m/s if the altitude is more than 20m *above takeoff point*, and then slows down to 0.5m/s until it touches the ground.

In our scenario, we need to land under SDK control on a hill that is more than 20 m higher than the takeoff point. This results in very hard landing when aircraft hits the ground at 3 m/s.

Is there any way to override this behavior ? We do know the altitude of the ground at the landing point, so it would be straightforward to provide it to the land command.

[SDK v3.2, M600PRO 1.0.1.20]

rohitsantdji, 2017-08-14:
Hi Vadim,

There is currently no way to have automatic landing do this for you - however, you can make this behavior happen by using movement control commands. You could call velocity control with say -3 m/s when your M600 is at a reasonable altitude, then reduce the velocity linearly as you approach the ground, and finally call the disarm command when the flight_status goes from 2 to 1.

flytrex-vadim, 2017-08-14:
Thanks Rohit, sounds like a great workaround.

Do you know if the algorithm that decides when to switch the flight_status from 2 to 1 is the same in automatic landing and when using the motion control ?

In manual flight it seems that that when auto-landing the ground  contact detection is less "conservative", and it shuts down the motors sooner than when landing manually.  When landing in windy condition that couple of seconds make a lot of difference.

rohitsantdji, 2017-08-16:
We've unfortunately not done any rigorous tests to compare the ground contact detection for movement control v/s auto landing.

closed 2017-08-16 by flytrex-vadim