dji-sdk/Onboard-SDK #164: Waypoint sample with MissionPlanner (M100 OSDK 3.2)

Martintl, 2017-08-15:
I have problems with starting waypoint mission by using file from MP. 
I have message "Invalid waypoint point data". 

I do the same thing as in example -> https://github.com/dji-sdk/Onboard-SDK/blob/3.2/osdk-wrapper/src/LinuxWaypoint.cpp
Only i change longitude , altitude , and latitude .

rohitsantdji, 2017-08-15:
What lat/lon/alt values are you inputing? Note that the values must be in radians.

Martintl, 2017-08-16:
@rohitsantdji Hi! I think i found my problem . It causes bc i set home position in simulator same as on foto , but function "getBroadcastData().pos" return lon(0.873999) lat(0.251656) . How can i fix it? 
[image]

Martintl, 2017-08-16:
@rohitsantdji Hello and thank you for response! I think i found my problem . That happens bc i dont know how to convert data from GPS ( long , lat .. ) to lon/ lat that i get by "api->getBroadcastData().pos" .

rohitsantdji, 2017-08-16:
The values you get from `getBroadcastData().pos()` are in radians; to convert them to the values you set in mission planner you need to do a radians-->degrees conversion (*180/pi). Also for waypoint values, you need to set altitude < 400 ft (~120m) since it is interpreted as a relative altitude above takeoff point.

rohitsantdji, 2017-08-23:
Closing since the issue seems to be resolved - if you have follow-ups, please re-open.

closed 2017-08-23 by rohitsantdji

semihyildirim, 2018-08-09:
I want to enter my own gps coordinates to waypoint example. I get an error as WAYPOINT_MISSION_INVALID_POINT_DATA. I  need to change in function of createWaypoints(line of 198 at Onboard-SDK/sample/linux/mission_sample.cpp). The function arguments provided by subscribeGPosition but I want to supply my own. How to do it?

nkkoufo, 2018-08-13:
hi semihyildirim, have you made sure that when you set the latitude and longitude for your waypoints that they are in radians? The documentation for dji says that it needs to be in radians you can convert from degrees by (pi * (latitude or longitude in degrees)) / 180

semihyildirim, 2018-08-15:
Yes, I have done it.