dji-sdk/Onboard-SDK #167: Do we have any way to know that the drone is being failsafe RTH?

magenious, 2017-08-18:
Hi, dear dji developers,
      As the title, in sdk, do we have any flag or api which let me know that the  failsafe was triggered and the drone was on the way to home point? I really need this information to hint the user when drone was RTH, thank you.

rohitsantdji, 2017-08-23:
Hi @magenious  - yes, you can subscribe to TOPIC_DISPLAYMODE and catch a transition to MODE_NAVI_GO_HOME. The API reference for Display Mode is [here](http://developer.dji.com/onboard-api-reference/namespaceDJI_1_1OSDK_1_1VehicleStatus_1_1DisplayMode.html#a56c83bf02178d81e17502ddde4eb17daa58189dd75592d1430d6b60a8ce9624a6).

magenious, 2017-08-24:
OK, that's exactly what i want, thanks a lot.

magenious, 2017-08-24:
Hi, @rohitsantdji 
I just realized that this flag could tell us whether the drone is returning home, but can not distinguish which kind of returning home. I suggest the SDK provide more information to distinguish smart RTH, low battery RTH and failsafe RTH, what do you think? Thank you.

rohitsantdji, 2017-08-24:
Today, you can distinguish low battery RTH with some logic in your code - monitor the battery voltage through TOPIC_BATTERY as well as the TOPIC_DISPLAYMODE. In the next firmware release we will be adding a flag to distinguish whether RC is connected or not.  When that happens, you should be able to distinguish between the three modes.