dji-sdk/Onboard-SDK #170: velocityAndYawRateCtrl doesn't apply the good velocities under an high voltage line

frederictedesco, 2017-08-21:
Hi,
I writed a software that autopilot a drone using the Onboard-SDK 3.3. Everything works well when I made some tests in my garden. But when I try to use my software under an high voltage line, everything works well except the velocityAndYawRateCtrl. The GPS signal is good.
Is somebody already has this kind of problem?
Thanks in advance for the help.
Regards,
Frederic

rohitsantdji, 2017-08-23:
Hi Frederic - good velocityAndYawRateCtrl depends greatly on having good magnetometer sensor readings. It is quite possible that the high-voltage line is interfering with the magnetometer.

frederictedesco, 2017-08-23:
Thanks for the answer. 
Do you have some ideas to protect the magnetometer from the electromagnetic waves?
Is the DJI Guidance system can solve my problem to apply the good velocities using the velocityAndYawRateCtrl function?

frederictedesco, 2017-08-23:
In the velocityAndYawRateCtrl's documentation, I read that "when Guidance system is working properly with Autopilot， horizontal velocity control（HORIZONTAL_VELOCITY）related control modes can be used." 
I'm using an A3 with a Rasrberry (UART communication). The guidance system will be connected to the raspberry using a USB port. How can I inform the A3 that a guidance system is working properly on the drone?

rohitsantdji, 2017-08-23:
Unfortunately, it is not possible to connect a Guidance system to the A3. The docs regarding guidance are related to Matrice 100 only.

frederictedesco, 2017-08-24:
Ok but how can I use the velocityAndYawRateCtrl function indoor without GPS and with no way to configure the A3 in "Guidance Autopilot"? 
Is it something I can do to bypass these requirements?

rohitsantdji, 2017-08-24:
There is no way to use velocity control indoors without GPS on an A3 - sorry!

closed 2017-09-07 by rohitsantdji