dji-sdk/Onboard-SDK #174: problem to connect the Raspberry Pi3 to the Matrice 100

stavshimko, 2017-08-28:
I am trying to connect a raspberry pi 3 as my onboard device onto my M100. I am connecting my pi by connecting the UART_CAN2 port to the GND, TXD, and RXD pins of my pi with the provided UART serial cable. I am using the 3.3 branch of the SDK provided by DJI on Github. In addition, if the drone was connected to mac the simulator is work and if connect to pi3 it's not working. 
please help me.
[image]

Okyara, 2017-08-28:
Hi there. We added M100 compatibility in release 3.3.1. Please check it out to ensure full support. You probably went over our  tutorial for connecting Raspberry Pi with M100 but just in case, take a look [here](https://github.com/dji-sdk/Onboard-SDK/wiki/Tips-on-connecting-the-Raspberry-Pi-to-the-Matrice-100). Let us know if the problem persists.

stavshimko, 2017-08-29:
it's didn't work. Im using GPIO serial port pins and my device is /dev/ttyAMA0 for Pi3, Otherwise it doesn't recognize.

rohitsantdji, 2017-08-29:
The `Port did not send` line from your screenshot strongly indicates that either the serial port is not correctly set up, or you do not have permissions to access the serial port correctly.

1. You could try exchanging the RxD and TxD pins and running the same sample again
2. Have you added yourself to the `dialout` group, then logged out and logged in as mentioned in the [checklist](http://developer.dji.com/onboard-sdk/documentation/quick-start/quick-start.html) here?

hdino, 2017-08-30:
This looks similar to [issue 158](https://github.com/dji-sdk/Onboard-SDK/issues/158) and [issue 142](https://github.com/dji-sdk/Onboard-SDK/issues/142).

After powering up the aircraft we receive data continuously from the flight controller (you can check using a simple terminal program like cutecom). Did you observe the same?

stavshimko, 2017-08-31:
[image]
still not working.
im afraid to replace the RXD and TXD pins, Is it dangerous?

stavshimko, 2017-08-31:
In Addition,
I can't opening the simulator if the drone connected to raspberry pi3

stavshimko, 2017-08-31:
In class setupOSDK I manage to connect to the drone because vehicle->protocolLayer->getDriver()->getDeviceStatus() is True but the function getFwVersion() return 0x0 and my raspberry pi3 is Ubuntu mate 16.04

hdino, 2017-09-06:
Usually it's not dangerous to swap the RX and TX pin. Indeed, often it's unclear from documentation whether the labels are meant to be seen from host or device side, so you cannot avoid doing try and error.

dxywing1314, 2017-11-03:
Hope you already solved this issue. If not, following suggestions may help. It seems that your pi3 does not receive anything from the drone. So it might be your pi3 serial port configuration problems.

1. disable the Bluetooth to allow /dev/ttyAMA0 access the UART port
2, go to /boot/cmdline.txt file, find console=ttyAMA0, and delete it to disable the UART port to be used for displaying the kernel information.
3. It is better to configure the UART port baud rate to 230400. You could go to /boot/config.txt and add init_uart_clock=64000000 to do so (I'm using pi2, did not test this on pi3).

closed 2019-05-28 by hummelslei