dji-sdk/Onboard-SDK #175: Should M600Pro 3.2.41.5 Flight Controller Version work with OSDK 3.2?

frontw, 2017-09-01:
Hi!
Should M600Pro 3.2.41.5 Flight Controller Version work with OSDK 3.2?
How could I downgrade Flight Controller firmware to use OSDK 3.2?
Thanks!

rohitsantdji, 2017-09-01:
You should be able to use DJI Assistant 2 to downgrade firmware to version 1.0.1.20, which supports OSDK 3.2.

frontw, 2017-09-01:
Hi, Rohit! 
Thank you.
I should have option to downgrade in Assistant Firmware menu?
[image]

frontw, 2017-09-01:
Downgrading through DJI GO right upper corner option. Thanks for answer about compatibility.

frontw, 2017-09-03:
Hi! 
It was my mistake - it is not option for main controller update in DJI Go, its about lightbridge.
So could you advice, please, how to downgrade m600pro controller firmware to work with 3.2OSDK. 
Thanks!

rohitsantdji, 2017-09-05:
Hi @frontw. If you absolutely need to get 3.2 firmware, we've set up a procedure for recovering the old firmware on a temporary basis, until you are able to migrate your existing code to OSDK 3.3. To access this procedure, please email dev@dji.com requesting an OSDK-related firmware downgrade.

The new 3.3 firmware brings safety updates regarding batteries and regulatory updates regarding no-fly-zones, so we recommend upgrading your codebase to 3.3 APIs as soon as possible.

closed 2017-09-07 by rohitsantdji