dji-sdk/Onboard-SDK #176: [M600 Pro][Error] input address range fault (11) when starting activation process

Zunhammer, 2017-09-07:
## Expected Behavior
Activation process running and succeding like described in documentation:
- clone and install DJI-OSDK
- install ROS with components and DJI-OSDK ROS
- create developer account, create app, etc.
- hardware setup (connections, etc.)
- perform "roslaunch dji_sdk sdk.launch" on onboard computer connected over serial to USB converter
- activation process running and succeeding

## Current Behavior
after running "roslaunch dji_sdk sdk.launch" I get the usual output first which is parameters, nodes, some text output about PID etc. Then the node tries to open the serial port and read data from it. Here i get the following error:

```
[...]
process[master]: started with pid [5581]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2a27d162-93bc-11e7-a1fd-00044b5adec1
process[rosout-1]: started with pid [5594]
started core service [/rosout]
process[dji_sdk-2]: started with pid [5608]

STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 115200...

STATUS/1 @ init, L47: ...Serial started successfully.
[ERROR] [1504782203.893107348]: Failed to read from serial device. The Onboard SDK is not communicating with your drone.
[ERROR] [1504782203.897830213]: Vehicle initialization failed
```
Obviously the serial can be used as it says serial started successfully. But I cant see any data on the serial lines (checked with digital analyzer).
A dmesg command gives the following output with the "input address range fault (11)" error of dji_sdk process:
```
[ 7459.119577] dji_sdk_node[5278]: unhandled input address range fault (11) at 0x206c6d783f490e, esr 0x92000004
[ 7459.119584] pgd = ffffffc07bf9f000
[ 7459.123031] [206c6d783f490e] *pgd=0000000000000000

[ 7459.127953] CPU: 0 PID: 5278 Comm: dji_sdk_node Not tainted 3.10.96-tegra #1
[ 7459.127958] task: ffffffc0f6452780 ti: ffffffc077418000 task.ti: ffffffc077418000
[ 7459.127965] PC is at 0x608564
[ 7459.127968] LR is at 0x575b78
[ 7459.127972] pc : [<0000000000608564>] lr : [<0000000000575b78>] pstate: 20000000
[ 7459.127975] sp : 0000007fc1b39310
[ 7459.127978] x29: 0000007fc1b39320 x28: 0000000000000000
[ 7459.127983] x27: 0000000000000000 x26: 0000000000000000
[ 7459.127988] x25: 0000000000000000 x24: 0000000000000000
[ 7459.127993] x23: 0000000000000000 x22: 0000000000000000
[ 7459.127997] x21: 0000000000000000 x20: 0000000000000000
[ 7459.128002] x19: 00000000006cdc70 x18: 0000000000000a03
[ 7459.128006] x17: 0000007f8eee01c0 x16: 0000007f8ef52120
[ 7459.128011] x15: 0000007f8ef81cc0 x14: 0000000000000000
[ 7459.128015] x13: 646f6d2f6372732f x12: 0000007f88f04190
[ 7459.128020] x11: 00000000003d0f00 x10: 0101010101010101
[ 7459.128025] x9 : 0000000000000000 x8 : 0000000000000065
[ 7459.128030] x7 : 0000000000000000 x6 : 0000007fc1b3927c
[ 7459.128034] x5 : 0000007f8c74f108 x4 : 0000007f8c74f000
[ 7459.128039] x3 : 0000000000000002 x2 : be62e867fab14a00
[ 7459.128044] x1 : be62e867fab14a00 x0 : 76206c6d783f3c0a

[ 7459.128056] Library at 0x608564: 0x400000 /usr/local/tor/ros_workspace/devel/lib/dji_sdk/dji_sdk_node
[ 7459.137289] Library at 0x575b78: 0x400000 /usr/local/tor/ros_workspace/devel/lib/dji_sdk/dji_sdk_node
[ 7459.146649] vdso base = 0x7f8ef7f000

```
It seems that there is some kind of wrong memory address by using the serial port? Maybe a null-pointer?
Anyway, I can use the serial port which is /dev/ttyUSB0 and write data on it manually. Also my user has all the required rights to use the serial port. So I think the problem is not related to the hardware.

## Your Environment

* Drone: DJI Matrice M600 Pro (FW: v1.00.01.60)
* Embedded computer: NVidia Jetson TX1
* Connection: USB to 3.3V TTL converter on /dev/ttyUSB0
* Operating System and version: Ubuntu 16.04 LTS
* ROS: ROS kinetic
* OSDK: v3.3.1

rohitsantdji, 2017-09-07:
Can you try to use a native serial connection (/dev/ttyTHS2 or similar) with the TX1? We've seen issues with USB-TTL converters on the TX1 specifically, the issue does not seem to exist on the TX2.

Zunhammer, 2017-09-08:
Thanks for your fast reply. As proposed I tried all the native serial ports (/dev/ttyTHS1 to 3) and I get the exact same output as posted before for all of them.

Additionally I tried the procedure on a Raspberry Pi 3 (running Ubuntu 16.4.2 Mate with ROS kinetic) to see if it is depending on the USB to serial converter. Actually, I don't get the "input address range fault" error here. Nevertheless, I still get the "Failed to read from serial device" message during initialization in ROS and cannot see any data sent out on the serial line.

Could you maybe give some information about what exactly should happen after the
`STATUS/1 @ init, L47: ...Serial started successfully.`
line? Should the drone be sending data all the time or should it start sending after the OSDK initiates the communication?

As the error says "Failed ro read from serial device" I'm wondering if the OSDK isn't able to access the serial device or if it is missing data on the Rx line? Could it be some kind of permission problem?

Thanks for your help!

Zunhammer, 2017-09-08:
Small update...
As I found here [http://forum.dev.dji.com/thread-33543-1-1.html](url) I can solve the "Failed to read from serial device." problem by commenting some lines in dji_sdk_node.cpp as follows:
`/*  bool setupStatus = validateSerialDevice(linuxSerialDevice);
  if(!setupStatus)
  {
    delete (linuxSerialDevice);
    return false;
  }
  else
  {
    delete(linuxSerialDevice);
  }
*/`
So now I see some data sent out on the serial port of Raspberry Pi (still need to test on Jetson TX1) and the process goes some steps further but fails at validatiing the Firmware Version:
```
STATUS/1 @ init, L47: ...Serial started successfully.

STATUS/1 @ init, L37: Attempting to open device /dev/serial0 with baudrate 115200...

STATUS/1 @ init, L47: ...Serial started successfully.

ERROR/1 @ parseDroneVersionInfo, L413: Drone version was not obtained. Please restart the program or call getDroneVersion

ERROR/1 @ getDroneVersion, L1031: Drone version not obtained! Please do not proceed.
Check if your serial connection is okay, and if your activation status is okay.

ERROR/1 @ functionalSetUp, L111: Failed to initialize Version! Please exit.
[ERROR] [1504867930.777077116]: Vehicle initialization failed
```

The M600 Pro is set to the same Baudrate (115200) and SDK is enabled with connected radio and app opened on mobile with internet access.

Due to the new behaviour and the data which is sent out after commenting the named lines, obviously there is a problem in validating the serial device. Maybe this helps in finding the actual problem?

Still I'm not able to activate the drone. Any ideas about the new status? As I can see on your website [http://developer.dji.com/onboard-sdk/documentation/appendix/versioning.html](url) my Firmware version (V1.0.1.60) should be compatible with OSDK 3.3.1.

Thanks in advance

Zunhammer, 2017-09-08:
Next update :)

after resetting and reinstalling DJI Assistant I was able to perform the activation process. I don't know if I did a mistake or if something else went wrong. Anyway, it even works on the Jetson using the USB to serial converter. However, I still see the the error in dmesg and had to do the workaround with commenting lines as described above. Maybe you can find a solution for that.

So my problem is solved for now. Thank you for your help!

closed 2017-09-08 by Zunhammer

pavankumarbn, 2018-04-24:
I am facing "failed to read from serial device. the on board sdk is not communicating with your drone." 
"Vehicle initialization failed". Matrice 100, Platform: Ubuntu 14.04LTS , ROS indigo, Onboard intel NUC computer. any suggestions how to resolve the issue.

Zunhammer, 2018-04-24:
Could you provide some more detailed information please.

How does your configuration look like? Is the serial port working in general (for example to an Arduino or another computer)?

pavankumarbn, 2018-04-24:
Thank you for the reply @Zunhammer 
Please find the informatio

... logging to /home/velab14/.ros/log/a5ca7520-4787-11e8-96a1-b8aeed721359/roslaunch-velab14-desktop-6169.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://velab14-desktop:46015/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1055128
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: 083ca2f1914c46e0c...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [6181]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a5ca7520-4787-11e8-96a1-b8aeed721359
process[rosout-1]: started with pid [6194]
started core service [/rosout]
process[dji_sdk-2]: started with pid [6201]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.
[ERROR] [1524550857.575794572]: Failed to read from serial device. The Onboard SDK is not communicating with your drone.
[ERROR] [1524550857.576145136]: Vehicle initialization failed

pavankumarbn, 2018-04-24:
@Zunhammer  do we need to give permissions to UART cable? can you please give a glimpse how to do it?

pavankumarbn, 2018-04-25:
Thank you for the suggestions.
1. There is no version variation. All are with compatible versions.
2. Yes user is a member of Dialout  group.
3. Cross verified. Rightly connected. 
4. I am using DROK DC Volt Converter Regulator to draw power from drone battery to onboard computer.
5. TTL- USB is working fine.
 
still it is giving same the problem. I could not able to figure out what is exact cause for this problem.

Zunhammer, 2018-04-25:
OK. I'm sorry that your problem still exists.

Depending on your knowledge in electronics I would suggest to anaylze the serial communication by using a digital analyzer or an oscilloscope to see what's happening and if BAUD rate and other configurations are working as desired.

Unfortunately, beside this I don't have any more advices for you. You should wait for someone of the DJI support team to answer as I'm only an end user like you.

Good luck!

pavankumarbn, 2018-04-26:
Thank you very much for your support @Zunhammer.  Which version of SDK do you suggest to use? currently I am using 3.4 ( SDK and ROSDK) version.