dji-sdk/Onboard-SDK #177: How could I use my own serial driver class in the framework of OSDK 3.3?

magenious, 2017-09-08:
As the image below shows, currently I wanna used my own implemented serial driver class in windows platform, and I want to keep the OSDK 3.3 code untouched, can we achieve this only by inheritance? I'm sorry I could find the solution, I just didn't find the interface that could accept my implemented serial driver object.
[image]

Well, I think OSDK 3.2 is more convenient to isolate the interface and implementation, I used my own serial driver as follows by 3.2:
[image]

Great appreciate for your help.

magenious, 2017-09-14:
Hi, Any updates?

rohitsantdji, 2017-10-11:
@magenious sorry for the delay - as you show in your first image, you should make your own HardDriver class inherit from DJI::OSDK::HardDriver, and your own ThreadManager (mutexes, condition variables etc.) inherit from DJI::OSDK::ThreadAbstract. You will also need to write your own threading class and inherit from DJI::OSDK::Thread, and initialize in dji_vehicle.cpp (function mandatorySetup).

magenious, 2017-10-12:
@rohitsantdji  Thanks for your comment. Yes, I need to write my own driver class and threading class, but even that i still couldn't let code of DJI SDK untouched, right? Just as the first of the image i posted above, i need to modify file dji_open_protocol.cpp for creating my own defined objects, I don't think this is a good idea, especially when i want to update sdk frequently.
Thank you.

rohitsantdji, 2017-10-12:
Yes, you are absolutely right - you will need to change code within some dji_open_protocol and dji_vehicle.

I would say that thew tradeoff between this design and the old one is such: for the new design, it is more accessible since developers just need to iinclude a single file `dji_vehicle.hpp` , instantiate a single object `new Vehicle(...)` and they're up and running. The old design has more components, so though it is harder to set up, you have easier configurability. From our end, the direction we want to go is to reduce the amount of time developers spend *within* the SDK (by writing threading models, platform drivers etc) and increase the time spent *using* the SDK APIs. For platforms that are not officially supported, such as Windows without Qt drivers, that makes it a little more cumbersome.

In the next major update, we are introducing a new `PlatformManager` abstraction that will consolidate all platform-specific intiializations within a single class, and make it easier to add your own to the SDK by just changing one file.

magenious, 2017-10-16:
OK, it's much clear to me now, but for me, I prefer to no modification. I had planed to compile the sdk as a library which could be included to my application, and more importantly, the sdk could be updated frequently just from github every time since it's none business with my modification. Now as v3.3 sdk needs me to modify one or two files, I have to include all the source files to my project and i have to  replace sdk files if i need to update the sdk, which is not convenient.
Do you have better idea? Thank you.

adrianhsm, 2017-10-16:
It is also a concern to me , what I want is that the dji sdk serve as a library, not something that i need to change the source code to support my appplication, what is worse, the code keeps changing, I need to merge my modification with the new code every time it changes, which seems to tedious to me, and add a heavy burden for code management  while I need to manage my code and dji code

rohitsantdji, 2017-10-16:
Thanks for the feedback. We're still evolving OSDK 3.3 and I think for the next couple of releases, you'll continue to see some more changes in the internal structure (the API remains the same). 

@adrianhsm the SDK can function as a library that you don't need to modify if you are using it on platforms we support. Unfortunately non-Qt Windows is not one of them at the moment.

magenious, 2017-10-18:
@rohitsantdji Thank you, hope we have better solution in the next release.

magenious, 2017-10-18:
Close the issue since the discussion is clear now.

closed 2017-10-18 by magenious