dji-sdk/Onboard-SDK #178: Waypoint uploads error: there is not enough memory.

szx0112, 2017-09-11:
I am running the waypoint demo on the upboard computer, when the number of waypoint is increased to 27, the error appear and says" ERROR, there is not enough memory". Then I take a look at the memory usage and find it only used 1000M/3.3G.

Any clue to solve this problem? Thanks.

justwillim, 2017-09-13:
this log means you send data frames too fast through OSDK pipeline, please try to send one waypoint data after the before one have been acked from the FC.
you could use a self-defined callback function to achieve this

zhongzhaoji, 2018-04-19:
I have two questions?Thanks.
  1.what is the "FC"?
  2.what can I do to know that sending waypoint data have been acked?

tahsinkose, 2019-02-07:
1. FC is Flight Controller.
2. In fact, `uploadIndexData()` of `wpMission` is a blocking call. Namely, it returns an ACK at each distinct waypoint upload. Furthermore, that ACK is explicitly tested [in here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/5325b657fe8020bf29f79944b80e0a2cc17623f1/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp#L105) to continue waypoint uploading operation. So, you don't need to do anything more (of course if you use Linux samples, then you can copy paste from ROS repo).

Nonetheless, I'm not sure why they have added sleep() calls [in here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/5325b657fe8020bf29f79944b80e0a2cc17623f1/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp#L117), even in this sequential scenario. I have removed them and tested up to 99 waypoints and saw no such error as inadequate memory.

@justwillim, if you have any further knowledge, please share with us. The API defines `uploadIndexData()` function as a blocking call, thus I can't see a reason for additional synchronization through callbacks. Moreover, I couldn't see any callback mechanism for waypoint uploading defined by OSDK. I believe it is an immediate functionality.

closed 2019-05-28 by hummelslei