dji-sdk/Onboard-SDK #179: Unexpected MISSION_IN_PROGRESS error and weird behaviour

talz, 2017-09-17:
I've been getting MISSION_IN_PROGRESS errors in some unexpected scenarios.

1. The drone has completed a waypoint mission in the air (without landing) and I'm uploading a new route while it's in the air.

2. Calling waypoint mission stop (after the drone has landed, or in the previous scenario before uploading a new route) - returns with this error.

3. After landing (manually or via the 'land' control command) and starting a new mission - calling the waypoint mission start will fail with this error - but the drone will take off non the less.

tahsinkose, 2019-02-07:
Hi @talz, I wonder did you find out any solutions?

talz, 2019-02-07:
No.
IIRC, I just teared down the connection to the drone, and reinitialized it.

tahsinkose, 2019-02-07:
As I just diagnosed, FC treats initialized _but not started_ missions as IN_PROGRESS. Started missions do have preemption logic; but solely initialized (i.e. uploaded) missions do not. That's why, one needs to restart SDK to preempt buffers.

closed 2019-05-28 by hummelslei