dji-sdk/Onboard-SDK #180: Support for m200 Series

a2hill, 2017-09-21:
Is there an approximate date for when we will be seeing OSDK support for the m200 series?

rohitsantdji, 2017-09-21:
We're looking at an OSDK release with M210/M210 RTK support by early October - around the second week. Hopefully it might go out even earlier!

rohitsantdji, 2017-10-16:
Update - We're putting the firmware and DJI Assistant 2's M210 simulation visualizer through testing, and we're holding the release of OSDK for M210/RTK until those go out. ETA is now first week of November.

AurelienBallier, 2017-11-13:
Hello,
Still any news from M210 support?
Regards,

amenonDJI, 2017-11-15:
@AurelienBallier @botaohu Apologize for the delay. We are now shooting tentatively for November 22nd.

AurelienBallier, 2017-11-16:
Thanks for the update.

plant99, 2017-12-14:
Hey @amenonDJI  updates?
Thank you.

rohitsantdji, 2017-12-14:
@plant99 We released OSDK 3.4 on Nov 22 with full support for M210/RTK, as well as some great new features. Get the[ latest 3.4.2 release](https://github.com/dji-sdk/Onboard-SDK/releases/tag/3.4.2) for some post-release bugfixes as well!

closed 2017-12-14 by rohitsantdji