dji-sdk/Onboard-SDK #182: Firmware compatibility issue between OSDK 3.3.1 and M100 3.1.10.0

jungwonkang, 2017-09-22:
Hi, I am using an Intel NUC computer(os: ubuntu 16.04) as a payload in M100.
I connected the Intel NUC to M100 using dji-recommended USB to TTL cable, 
and I've tried to run a dji-provided linux sample program.
In the DJI Assistant 2, I checked 'Enable API Control'.
Then, I just followed the procedure in 'Linux OES' in the webpage:
https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html

However, when I ran the sample using the following command as in the webpage.
  $ ./djiosdk-flightcontrol-sample UserConfig.txt

I got the following errors(I am also attaching the screenshot):
  ......
  STATUS/1 @ init, L45: ...Serial started successfully.
 
  STATUS/1 @ parseDroneVersionInfo, L525: Hardware = 

  STATUS/1 @ parseDroneVersionInfo, L526: Firmware = 3.1.10.0

  STATUS/1 @ parseDroneVersionInfo, L529: Version CRC = 0x0
 
  ERROR/1 @ initVersion, L317: Your firmware is not supported by this version of the OSDK.
 
  ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
  Vehicle not initialized, exiting.
  ......

In my Intel NUC, I installed DJI Onboard SDK (OSDK) 3.3.1.
The firmware of M100 seems 3.1.10.0.
However, the firmware compatibility information in https://github.com/dji-sdk/Onboard-SDK
indicates OSDK 3.3.1 supports M100 (3.1.10.0).

Could you give me any comments for this issue?

Thank you.

[image]

zliDJI, 2017-09-23:
How did you get the OSDK? Did you do 
```
git clone https://github.com/dji-sdk/Onboard-SDK.git
```
or you downloaded the zip file? Please try to do it through `git clone`.

rohitsantdji, 2017-10-11:
Closing due to inactivity, since @zliDJI's solution should work.

closed 2017-10-11 by rohitsantdji