dji-sdk/Onboard-SDK #185: SDK readPoll thread using 100% of CPU

sergio303, 2017-10-04:
Hi,

I'm succesfully using latest Onboard-SDK with a Matrice 100, but I see a 100% cpu usage as soon as the serial communication is started:

[image]

This is the output from the app:
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L37: Attempting to open device /dev/serial0 with baudrate 230400...
STATUS/1 @ init, L47: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L568: Device Serial No. = xxxxxxxxxxx
STATUS/1 @ parseDroneVersionInfo, L570: Hardware = M100
STATUS/1 @ parseDroneVersionInfo, L571: Firmware = 3.1.10.0
STATUS/1 @ parseDroneVersionInfo, L574: Version CRC = 0xA6453AAC
STATUS/1 @ functionalSetUp, L126: Subscriber not supported!
STATUS/1 @ functionalSetUp, L164: MFIO not supported!
ERROR/1 @ functionalSetUp, L182: Hardware Sync not supported!
STATUS/1 @ activate, L951: version 0x3010A00
STATUS/1 @ activate, L964: Activation successful
…

from there on I only read telemetry data, but CPU is 100% on readPoll thread as long as I'm using the API.

Is there any way to avoid this?

Thanks!

rohitsantdji, 2017-10-10:
Hi Sergio, could you tell us what onboard computer you are using? I see that you are connecting to the drone at 230400 baud, and you have an M100, so the max telemetry frequency is going to be 100 Hz - in which case you can safely increase the sleep time on [this](https://github.com/dji-sdk/Onboard-SDK/blob/3.3/osdk-core/platform/linux/src/posix_thread.cpp#L126) line to something of the order of 100 us rather than 10 us. That should help reduce CPU usage.

sergio303, 2017-10-11:
HI, I'm using a single core Raspberry Pi Zero, hence the need to free some CPU for other uses. I've increased the usleep time and that seems to free up enough CPU for my needs. Thanks!

closed 2017-10-11 by rohitsantdji