dji-sdk/Onboard-SDK #187: How to get  timestamp on the Flight Controller?

botaohu, 2017-10-10:
In all recorded video by DJI drones, it has a timestamp (UTC datetime) for creation time.

But from OSDK, how can I get the current time in UTC of the flight controller?

The timestamp from broadcast data only contains "time: uint32, times_ns: uint32"
But in this structure, the time is the duration after last time the flight controller power cycled.  
We cannot get the absolute time of the current flight controller.

sergio303, 2017-10-10:
You could try getting time from the Telemetry::GPSInfo structure (not supported on Matrice 100).

zliDJI, 2017-10-10:
Sergio is right.

sdo91, 2018-03-12:
@sergio303 @zliDJI can you expand on getting time from this Telemetry::GPSInfo structure? Is there a topic I can subscribe to to get this data?

closed 2019-05-28 by hummelslei