dji-sdk/Onboard-SDK #188: [Announcement] M600/Pro Firmware Downgrade Advisory

rohitsantdji, 2017-10-12:
[Mirror of dji-sdk/Onboard-SDK-ROS#146]

Hi all,

Over the last few days, we have received a small number of reports regarding the in-flight stability of the M600/Pro 1.0.1.60 firmware. We're currently investigating this issue, but in the meanwhile we've pulled the firmware from DJI Assistant. Here's how this might affect you:

- If you are already on 1.0.1.60, you will be prompted to unlock the drone each time from the Go app before you fly. You will be able to downgrade firmware to 1.0.1.20.
- If you are on 1.0.1.20, you can continue to use OSDK 3.2 with this firmware. You will not be given an option to upgrade to 1.0.1.60.

We realise that the change of APIs for OSDK 3.2 --> 3.3 means this downgrade advisory is a problem for your code, so to mitigate this we're working on a version of OSDK 3.3 that will be compatible with the 1.0.1.20 firmware. This version will keep the same interface and APIs as OSDK 3.3, but newer features such as high-frequency telemetry and hardware sync will be unavailable. This version should be out by the end of next week. We apologize for the churn in firmware, and hopefully this backward compatible SDK will go some way towards ensuring everyone is flying a flight-worthy firmware without much change in your codebase.

rohitsantdji, 2017-10-19:
Hi everyone - M600 FW 1.0.1.20 can now be used with the latest OSDK 3.3 release and OSDK ROS 3.3 release. Thank you for your patience!

closed 2017-10-19 by rohitsantdji

eagleman111, 2017-10-20:
Hi, how to go back to 1.20? 1.60 make alot of troubles

rohitsantdji, 2017-10-20:
@eagleman111 - you can downgrade through DJI Assistant 2.

To help us understand your issues better, could you tell us some specifics on 1.60 that is causing trouble?

eagleman111, 2017-10-20:
flight instability.
I cant see in DJI Assistant 2 a downgrade button?

snehagn, 2017-10-20:
I tried the updated SDK and ROS SDK 3.3 for 1.0.1.20. I am getting a run error of [dji_sdk-2] process has died [pid *pid*, exit code -11, cmd /path/to/folder/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/path/to/log/file.log]. Is there a way to fix this without having to clean my system?

rohitsantdji, 2017-10-20:
@eagleman111 
[image]

This is what you see with Assistant 1.1.6. Could you report back with what you see?

eagleman111, 2017-10-21:
worked. however there is no more notification of 1.60. is this normal?

rohitsantdji, 2017-10-26:
M600/Pro firmware 1.0.1.65 was released on 10/24, this fixes the instability observed in 1.0.1.60. It is otherwise identical to the 1.0.1.60 firmware, and all code should work seamlessly across the transition.

eagleman111, 2017-10-27:
Guys, I have a big problem,
I was on 1.20, and everything is stable, I upgraded to 1.60, alot of instability issues, Then I downgraded to 1.20 based on your answer above, and now I am facing 2 problems:
1- After downgraded to 1.20, it is now not stable at all, how to make 1.20 stable again?
2- In DJI assistant 2, I cant see neither 1.60 nor 1.65 upgrades.
Any help please?

rohitsantdji, 2017-10-27:
Please hit the button in the top right corner for Assistant 2, that will help you refresh the firmware list. Only the 1.65 firmware should be seen. If that is not the case, please attach a screenshot and we'll help you resolve the issue.

1.20 firmware has not changed at all since the last time it was available, so it is likely that the root cause of your instability is lying elsewhere.

eagleman111, 2017-10-27:
so how to fix root issue?

eagleman111, 2017-10-31:
no help?
I upgrade to 1.65 but still same problems.
Any idea how to downgrade to 1.20

DamesNYC, 2017-11-01:
is it possible to downgrade from 1.65 to 1.20? it does seem to offer it as an option with assistant 2 (v1.16)

jtrovato, 2017-11-22:
Hello, 

I was recently forced into upgrading my M600. I believe the DJI Go app or controller was not letting me fly until I did. 

I upgraded to what I thought was the most recent about a week or two ago so V1.00.01.65, and it now lets me fly the drone in P-mode. Unfortunately my onboard computer with OSDK 3.2 for ROS is totally messed up. The SDK activates successfully but all of the messages are corrupt. The positions coming in, the rc_channels, pretty much everything has incorrect values. 

Now, I would like to downgrade to 1.20 but Assistant 2 v1.17 has no options to downgrade. Please let me know what the best way to downgrade is?