dji-sdk/Onboard-SDK #190: Flightcontrol sample crashes throwing CONTROLLER_FAIL

chocksmith, 2017-10-16:
Hello,

* The Assistant connects to the drone
* I can pilot the drone on the simulator using the remote controller and the DJI GO App
* The djiosdk-flightcontrol-sample compiles and run, showing the prompt

But when I select one of the two options of the flightcontrol-sample (a or b), it crashes:

    pi@drone:~/drone/Onboard-SDK-3.3/build/bin $ ./djiosdk-flightcontrol-sample ../../sample/linux/common/UserConfig.txt 
    Read App ID
    User Configuration read successfully. 

    STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

    STATUS/1 @ init, L47: ...Serial started successfully.

    STATUS/1 @ parseDroneVersionInfo, L568: Device Serial No. = 041DE52780

    STATUS/1 @ parseDroneVersionInfo, L570: Hardware = M100

    STATUS/1 @ parseDroneVersionInfo, L571: Firmware = 3.1.10.0

    STATUS/1 @ parseDroneVersionInfo, L574: Version CRC = 0xA6453AAC

    STATUS/1 @ functionalSetUp, L126: Subscriber not supported!

    STATUS/1 @ functionalSetUp, L164: MFIO not supported!

    ERROR/1 @ functionalSetUp, L182: Hardware Sync not supported!

    STATUS/1 @ activate, L951: version 0x3010A00

    STATUS/1 @ activate, L964: Activation successful
    | Available commands:                                            |
    | [a] Monitored Takeoff + Landing                                |
    | [b] Monitored Takeoff + Position Control + Landing             |
    a

    STATUS/1 @ getErrorCodeMessage, L590: dT?~???vT?~???v
    STATUS/1 @ getCMDIDTaskMSG, L762: CONTROLLER_FAIL

    STATUS/1 @ getErrorCodeMessage, L590: 
    STATUS/1 @ getCMDIDTaskMSG, L762: CONTROLLER_FAIL

rohitsantdji, 2017-10-16:
I see you're using a Raspberry Pi. We've seen reports that it doesn't behave as expected, and there are a few Github issues regarding this already. Maybe you could look through #135 , and the [wiki](https://github.com/dji-sdk/Onboard-SDK/wiki/Tips-on-connecting-the-Raspberry-Pi-to-the-Matrice-100) page here?

chocksmith, 2017-10-17:
Thank you for your help.
Yes, I am using Raspberry Pi 3b.
Before buying the M100 I contacted the support and asked if I could use it for OSDK and the answer was “yes”. Moreover, I haven’t seen any explicit indication that it would not work. On the other hand, documentation says that it works on ARM.
It surprises me that it fails so explicitly on the very first stage on the very first tutorial. Have it ever worked for someone? Or something is tremendously wrong on my environment?
Any clue?

chocksmith, 2017-10-17:
@rohitsantdji I visited the wiki page and the #135 and I am doing everything they recommend. It connects, but it throws the CONTROLLER_FAIL. Thank for your piece of advice.

I would appreciate more help guys. I am stuck here.

chocksmith, 2017-10-17:
Update: It worked!
I do not know exactly what I did to make it work. Things I did:

* Updated the iOS version
* Rebuild the project several times
* Reinstalled Assistant
* Reinstalled DJI Go

closed 2017-10-17 by chocksmith