dji-sdk/Onboard-SDK #191: Bug at flight_control_sample.cpp

chocksmith, 2017-10-17:
Hello,

This is the code that checks if the motor has started:

                while ((vehicle->broadcast->getStatus().flight
				!= DJI::OSDK::VehicleStatus::M100FlightStatus::TAKEOFF)
				&& motorsNotStarted < timeoutCycles) {
			motorsNotStarted++;
			usleep(100000);
		}

		if (motorsNotStarted == timeoutCycles) {
			std::cout << "Successful TakeOff!" << std::endl;
		}

The code above works, but it is not correct. The correct would be something like:

               while ((vehicle->broadcast->getStatus().flight
				< DJI::OSDK::VehicleStatus::M100FlightStatus::TAKEOFF)
				&& motorsNotStarted < timeoutCycles) {
			motorsNotStarted++;
			usleep(100000);
		}

		if (motorsNotStarted < timeoutCycles) {
			std::cout << "Successful TakeOff!" << std::endl;
		}

Same mistake happens on others subsequent checks.

rohitsantdji, 2017-10-17:
Good catch! Can you submit a PR?

chocksmith, 2017-10-18:
I will!
Tks!

commit 00a3afab (2017-10-19): Hotfix release adding 3.3 support for M600 FW 1.0.1.20. See #188. Misc bugfixes, closes #191 and some Qt sample issues.
closed 2017-10-19 by rohitsantdji