dji-sdk/Onboard-SDK #199: Problem after upgraded M60Pro from 1.20 to 1.60 and 1.65! Help!

eagleman111, 2017-11-01:
Hi,
I have a M600Pro with firmware 1.20 and it was working without any problem.
I recieved a message to upgrade ito 1.60, I did upgrade it to 1.60 the M600Pro starts having inistablity issues, I asked you and you told me to downgrade to 1.20, I did downgraded to 1.20, problems fixed, then you released version 1.65 indicating that the inistability issues have fixed, I upgraded to 1.65, the aircraft shown same inistability issues as 1.60.
Please help me either fix the inistability issues or a way to get bacl to 1.20 thanks

The weired behavior is that, when I power it on, and go for few meters, a message of "IMU heading exception in flight please switch to Atti mode", and  the aircraft show GPS signal very low, and switched to Atti mode, so I have to manually control it and land it.
Please give me go back 1.20

Aircraft is M600Pro
When I do tests now, There is no anypayloads. attached to it, as I am afraid of crashing

amenonDJI, 2017-11-17:
@eagleman111 It would be best if you can send us the logs from the Aircraft when you see this issue.

amenonDJI, 2017-11-22:
@eagleman111 Could you please contact support@dji.com? This seems to be a Aircraft specific issue.

closed 2019-05-28 by hummelslei