dji-sdk/Onboard-SDK #201: Matrice 100 with Raspberry Pi, Onboard-SDK Sample Code Failed and Drone Fell from Sky

dxywing1314, 2017-11-03:
I wrote my code based on Onboard-SDK Linux sample code flight-control. The simulation is OK and the results match my goal. However, for the actual test, the drone suddenly fell from the sky after it took off (thanks to the 1.2m takeoff altitude so the drone did not damage). 

In order to address the problem, I changed to the Onboard-SDK Linux sample code djiosdk-flightcontrol-sample. However, the drone still fell and behaved exactly the same. It seems that the drone lost the throttle when altitude command was sent. 

Another issue is that it seems that control priority did not hold when my controller is in F-mode. When I ran the sample code the drone fell to the ground, it still tried to roll for y-axis movement (seems planar movement is OK), so I pressed down the joystick firmly to force the drone landing. The drone should immediately act to land, but it still tried to perform y-axis movement until I changed the controller to P mode. 

My code can be checked at https://github.com/dxywing1314/Onboard-SDK/tree/3.3/sample/linux/flight-control

In all, the simulation is OK for both my code and sample code, but the actual test failed.

jswoboda, 2017-11-16:
I've had a similar problem with the ROS SDK. I would give it waypoints, which would then be interpreted through the OSDK, and the drone would fall from the sky. We tried it twice, luckily our operator took over just in time and landed the drone properly. I've noticed that the altitude I get back when I'm polling GPS coordinates, does not correspond to any reasonable altitude level either. This seems like its something to do with the way they're calculating their altitude.

I'm working with a Matrice 600 with a total weight of 31 lbs (14 kg). Beyond this **BEING A MAJOR SAFETY ISSUE** we almost lost a fair amount of expensive RF hardware. This really needs to be fixed ASAP.

amenonDJI, 2017-11-16:
@dxywing1314 @jswoboda Could you tell me what was the FC Firmware when you saw this behavior?

jswoboda, 2017-11-16:
I was using 1.0.1.20 and SDK 3.3.

dxywing1314, 2017-11-17:
@amenonDJI My flight controller version is 1.7.80. Aircraft version 1.3.1.00

amenonDJI, 2017-11-17:
@jswoboda Could you send me the logs related to the crash?

dxywing1314, 2017-11-20:
@amenonDJI Like I said FC firmware version is 1.7.80. I checked it from DJI GO app -> Settings -> About page. Any problems with this?

amenonDJI, 2017-11-20:
@dxywing1314 I apologize, I got mixed up with the M100 and M600 issue on the same thread. Thanks for the information. 
@jswoboda Could you please create a new issue with all the details on this thread. It would be easier to track that way.

dxywing1314, 2017-11-20:
@amenonDJI So any solutions to this problem?

jswoboda, 2017-11-21:
I was using the ROS API, should I create the issue there? Also what logs should I send you?

amenonDJI, 2017-11-22:
@jswoboda Yes, please create the issue under the ROS repo on a new thread. That would be Flight Controller logs - follow steps below to download logs: 

1.)  Go to Main Controller Settings on DJI Go
2.) Go to Advanced Settings
3.) Enter Flight Data Mode 

Now you can use MicroUSB to USB cable connected to your computer to see the log files. If you have flown the aircraft or started the aircraft several times after the behavior that you noticed, it is possible that the logs were re-written, in which case it would be harder to identify the problem. 

Were you ROS bagging any data when you noticed this behavior? If so, we may also be able to use this data to debug the problem.

amenonDJI, 2017-11-22:
@dxywing1314 We have not noticed this issue from our end - again, I would recommend extracting logs from the aircraft so we can identify the problem. Please follow steps from my previous post to extract logs.

jswoboda, 2017-11-28:
Hi @amenonDJI I'm downloading the logs right now and will open up a new issue in ROS. Before I do that what would be the best way of sharing the files for you?

L23J, 2017-12-28:
Hi,
By any chance, is there a resolution for this issue? I am experiencing something quite similar where the Matrice 100 falls to the ground after executing a modified version of the Onboard-SDK-3.4 Linux sample code flight-control. My setup is using a Matrice 100 and the DJI Manifold running Ubuntu Linux 14.04 with the following info from DJI Go App -> General Settings -> About:

App: 3.1.23
Version: 1.3.1.00
Remote Controller: 1.9.2

I assume the N1 flight controller FW is 3.1.6.103 seen below:

[image]

It doesn't look like I have any other FW to upgrade from the DJI Assistant 2 list:

[image]

Using Onboard-SDK-3.4:
I have only modified this section of the code -
 **moveByPositionOffset(vehicle, 1, 0, 1.22, 0) and commented out moveByPositionOffset(vehicle, 6, 0, -3, -30) and moveByPositionOffset(vehicle, -6, -6, 0, 0) as seen in the attachment below:**

[image]

Simulated results are promising with no reports of crashes and executes as planned. However, in real life tests, the Matrice 100 takes off, hovers in the air, and then crashes to the ground from ~1.2 meters in the air with the following task timeout error:

[image]

Are the FW versions listed above for the Matrice 100 correct for running the OSDK-3.4 or do I need an upgrade to one or more devices? More specifically, does the N1 controller FW info look to be correct for running the OSDK-3.4? 

Thanks in advance.

amenonDJI, 2018-01-04:
This is a bug and is being tracked at #225. Please do not use Flight Control sample till we have a fix.

closed 2018-01-04 by amenonDJI

J23L, 2018-01-05:
Hi @amenonDJI  - Thanks for the update. Do you know when a fix is expected?

Thanks again.

justwillim, 2018-02-06:
Fixed in 3.6