dji-sdk/Onboard-SDK #206: N3 not responding to Input given to "dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition" Node

iLenoy, 2017-11-11:
We are trying to control our N3 by giving the "dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition" the Joy message with a roll pitch yawrate and z position but it is not functioning correctly. We had it work 2 trys out of 40 so we are fairly certain that the control array we are publishing is the right format and contains the right data. We are using a Ubuntu 16.04 system connected via an USB-TTL which is connected as shown in the manual.
We are just unsure of why it only works 2 out of 40 trys. Is there a special sequence in which you have to Arm the Drone and contect the USB-TTL conector? Or are we missing something else or might it just be a hardware error for example the USB-TTL having a loose connection. Does anyone have experience with this ?

amenonDJI, 2017-11-22:
@iLenoy Could you share more information such as code snippet, frequency at which commands were sent, simulation/real world, OSDK version, FC version.

amenonDJI, 2017-11-22:
Closed. Duplicate thread.

closed 2017-11-22 by amenonDJI