dji-sdk/Onboard-SDK #211: How could we get firmware package version?

magenious, 2017-11-21:
As the image below shows, how could i get firmware package version by api in OSDK v3.3 ? It's tricky that the api Version::FirmWare Vehicle::getFwVersion() returns the Flight Controller Version, not the firmware package version.
[image]

Thanks.

rohitsantdji, 2017-11-29:
The firmware package number is seen in DJI Assistant 2's upgrade page. It is mentioned on this page as an alternative means of finding out your FW version, without first having to run OSDK. If you have a certain FC firmware version, it is guaranteed to correspond to the firmware package version mentioned in the same row in this table.

There is no way to get this version via API, since it does not offer any extra information over the `getFwVersion` API.

magenious, 2017-12-04:
Thanks or your confirm.

closed 2017-12-04 by magenious