dji-sdk/Onboard-SDK #213: the received data frame for callback of takeoff()/land()/gohome() is not correct

magenious, 2017-11-27:
Hi, dji developers,
         I used the following code to let the drone take off：
`pVehicle->control->takeoff(TaskCallback);`
and my callback function TaskCallback was defined as:
`void TaskCallback(
	Vehicle* pVehicle,
	RecvContainer recvFrame,
	UserData userData)
{
	string type = "task";
	string info = "fail to execute task";
	EExecStatus result = ERR;

	ACK::ErrorCode ack;

	if (recvFrame.recvInfo.len - Protocol::PackageMin <= sizeof(uint16_t))
	{
		ack.info = recvFrame.recvInfo;
		ack.data = recvFrame.recvData.ack;

		printf("cmd_set = %x, cmd_id = %x, ack.data = %x\n", ack.info.cmd_set, ack.info.cmd_id, recvFrame.recvData.ack);
		std::cerr << "cmd_set = " << ack.info.cmd_set << ", cmd_id = " << ack.info.cmd_id << std::endl;

		if (ACK::getError(ack))
		{
			GetErrorCodeMsg(ack);
		}
		else
		{
			result = SUCC;
			info = "task is executed successfully";
		}
	}
	else
	{
		DERROR("ACK is exception, sequence %d\n", recvFrame.recvInfo.seqNumber);
	}

	std::string sStr = ConvertToJsonStr(type, result, info, 0, 0);
	DyUtils::SaveLogInLocal(sStr);
	PushItemToList(sStr, listInfo);
}`
well, then I found that although the drone took off successfully in simulation , the value of recvFrame.recvData.ack was 0xcc00 in my callback function, which was not correct as the image showed below. It should be 0x0002 according to your open protocol.
[image]
[image]

I doubt it was the problem of firmware cause the other callbacks i used has no issue except for takeoff()/land()/gohome(), please help to check, thank you.

rohitsantdji, 2017-11-29:
Are you using an A3-based system? In that case, the open protocol doc is no longer valid for acknowledgements; see the [API reference](https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1ErrorCode_1_1ControlACK_1_1Task.html) for the most up-to-date error codes and ACKs.

In that context, there is a small bug in your callback; just like the default `actionCallback` in `dji_control.cpp`, you need to do `ack.data = recvFrame.recvData.commandACK` instead of `ack.data = recvFrame.recvData.ack`. The difference is that `commandACK` is a single byte ACK while `ack` is a two-byte ACK. Then, the value will be 0x00, which matches the SUCCESS value you expect.

magenious, 2017-12-01:
@rohitsantdji Yes, I used A3 pro in my project.  So as you comment, I should use recvFrame.recvData.commandACK instead of ack.data = recvFrame.recvData.ack to parse the ack value in callback function, but why? I'm sorry I haven't found the code that support what said yet, I just found that the recvData should contain the ack as in `Protocol::appHandler`:
[image]
As the `ACK::TypeUnion` has several ack types as below, would you please help to pointed out the code in sdk which determined which type of ack I should use, thanks.
`  typedef union TypeUnion {
    uint8_t  raw_ack_array[MAX_INCOMING_DATA_SIZE];
    uint8_t  versionACK[MAX_ACK_SIZE];
    uint16_t ack;
    uint8_t  commandACK;
    uint8_t  missionACK;
    uint8_t  subscribeACK;
    uint8_t  mfioACK;

    /*
     * ACK(s) containing ACK data plus extra payload
     */
    HotPointStartInternal    hpStartACK;
    HotPointReadInternal     hpReadACK;
    WayPointInitInternal     wpInitACK;
    WayPointDataInternal     wpDataACK;
    WayPointVelocityInternal wpVelocityACK;
    MFIOGetInternal          mfioGetACK;

    /*
     * Push Data in ground-station mode
     */
    WayPointReachedData wayPointReachedData;

  } TypeUnion; // pack(1)`

magenious, 2017-12-01:
Also, I found in sdk, the default callback for takeoff used ask:
`void
Control::takeoff(VehicleCallBack callback, UserData userData)
{
  this->action(FlightCommand::takeOff, callback, userData);
}`

[image]

`void
Control::actionCallback(Vehicle* vehiclePtr, RecvContainer recvFrame,
                        UserData userData)
{
  ACK::ErrorCode ack;
  Control*       controlPtr = vehiclePtr->control;

  if (recvFrame.recvInfo.len - Protocol::PackageMin <= sizeof(uint16_t))
  {

    ack.info = recvFrame.recvInfo;
    ack.data = recvFrame.recvData.ack;

    if (ACK::getError(ack))
    {
      ACK::getErrorCodeMessage(ack, __func__);
    }
  }
  else
  {
    DERROR("ACK is exception, sequence %d\n", recvFrame.recvInfo.seqNumber);
  }
}`

rohitsantdji, 2017-12-01:
> the default callback for takeoff used recvFrame.recvData.ask:

You are referring to a very old commit, this bug was fixed four months ago in OSDK 3.3.1, in commit b2f28f8.

> As the ACK::TypeUnion has several ack types as below, would you please help to point out the code in sdk which determined which type of ack I should use

I would say that the best way to understand the return type to use would be through the default callback implementation for specific APIs, or the corresponding blocking API call. All these calls are using the right data type from the TypeUnion for their specific purpose.

magenious, 2017-12-04:
@rohitsantdji OK, greatly appreciate for your feedback, thanks.

closed 2017-12-04 by magenious