dji-sdk/Onboard-SDK #214: [Mission can't start]

nguyenvietvu, 2017-11-29:
I connect M600 with Pi 3. every setup is ok.
- Onboard sdk 3.3

My home: {0.651475, 2.424084}
My mission[] (with latitude, longitude) is:
double missions1[5][2] = {
     {0.651477, 2.424084},
     {0.651477, 2.424085},
     {0.651476, 2.424085},
     {0.651475, 2.424084},
     {0.651476, 2.424083}}; -># run waypoint mission ok

But:
  double missions2[5][2] = {
      {0.651473, 2.424083},
      {0.651474, 2.424094},
      {0.651462, 2.424099},
      {0.651452, 2.424080},
      {0.651463, 2.424079}}; -> can't start with error: # MISSION_TOO_FAR_FROM_HOME

with my calculate, distance between home and farthest waypoint about 200 meters.
Please help me check: dpi_open_protocol.cpp:
void
Protocol::send(uint8_t session_mode, bool is_enc, const uint8_t cmd[],
               void* pdata, size_t len, int timeout, int retry_time,
               bool hasCallback, int callbackID)
{
  Command  cmdContainer;
  uint8_t* ptemp = (uint8_t*)encodeSendData;
  *ptemp++       = cmd[0];
  *ptemp++       = cmd[1];

  memcpy(encodeSendData + SET_CMD_SIZE, pdata, len);
  printf("SET_CMD_SIZE %d", SET_CMD_SIZE);
  cmdContainer.sessionMode = session_mode;
  cmdContainer.length      = len + SET_CMD_SIZE;
  cmdContainer.buf         = encodeSendData;
  cmdContainer.cmd_set     = cmd[0]; // cmd set
  cmdContainer.cmd_id      = cmd[1]; // cmd id
  cmdContainer.retry       = retry_time;

  cmdContainer.timeout = timeout;
  cmdContainer.encrypt = is_enc ? 1 : 0;

  //! Callback
  cmdContainer.isCallback = hasCallback;
  cmdContainer.callbackID = callbackID;

  sendInterface(&cmdContainer);
}

I think have problem in this function!

rohitsantdji, 2017-11-29:
Please try this debugging procedure:

1. In simulation, set your home point to {0.651475, 2.424084}
2. Connect DJI Go to your RC and make sure the internet is functioning on your mobile device
3. Using the RC, make the drone takeoff in simulation. This should update the homepoint set within the flight controller.
4. Now, retry your code. 

Do let us know the results of this procedure, and if it's still not working we can help you debug further.

nguyenvietvu, 2017-11-30:
Awesome, It's worked! thank for your support :)

p/s: but I don't understand why: missions1 is ok, missions2 is error, could you please explain me!

double missions1[5][2] = {
{0.651477, 2.424084},
{0.651477, 2.424085},
{0.651476, 2.424085},
{0.651475, 2.424084},
{0.651476, 2.424083}}; -> run waypoint mission ok

double missions2[5][2] = {
{0.651473, 2.424083},
{0.651474, 2.424094},
{0.651462, 2.424099},
{0.651452, 2.424080},
{0.651463, 2.424079}}; -> can't start with error: MISSION_TOO_FAR_FROM_HOME

rohitsantdji, 2017-11-30:
Good to know it worked! The debugging procedure is doing two things:

1. Setting the homepoint in the drone's internal flight controller
2. Setting the homepoint in the DJI Go App

There are some layers of logic that will throw the TOO_FAR_FROM_HOME error in various cases; when you are outdoors with non-simulated GPS signal, you will satisfy all the conditions and won't run into this error. However in simulation, you need to ensure that the homepoints are set in this manner, and the location you are simulating in Assistant is close to the location specified in the mission. Could you try a fresh run with the debugging procedure and just the second set of waypoints?

SpencerHehl, 2018-02-03:
Hi there, I am having this issue even outside the simulator with a waypoint between 60 and 100 meters away. Is there some sort of setting I need to configure to allow the Matrice 600 to fly farther?

rohitsantdji, 2018-02-05:
@SpencerHehl Please check whether your DJI Go app has turned on Beginner mode after an update. If that doesn't work, could you give us some more detailed info about the error you are facing? Some helpful things to include would be the error messages, the behavior in the simulator, and what you see on DJI Go when this issue happens.

SpencerHehl, 2018-02-05:
@rohitsantdji So I am not seeing a "Beginner" mode but I am seeing a "Coach" mode which I was assuming was the same thing. That is set to off. We also saw that there was a set max distance which we have tried disabled and enabled and set to 5000m and the behavior remains unchanged. Simulator gives us the same error which your above advice helped us one time but the rest of the time same issue. MISSION_TOO_FAR_FROM_HOME i believe is the specific error. We are using the ROS OSDK and specifically uploading waypoint missions. Both in simulator and in real life we are able to upload the points but not run the mission (it throws the error after saying it uploaded them successfully). Again we have run successful simulated and real tests with a range of about 30-40 meters from home, but then roughly 60+ is when we get the error. I am running these more or less based on the sample waypoint mission provided so I dont know if there is some setting or configuration elsewhere i need to set to unlock greater distances (beyond that beginner mode) but i'm not seeing where to find that information.

closed 2019-05-28 by hummelslei