dji-sdk/Onboard-SDK #216: Negative altitudes?

rpici, 2017-11-29:
Hello,

In OSDK 3.4, I'm running the linux telemetry sample, and I'm getting large negative values for altitude (`TOPIC_ALTITUDE_FUSIONED`).

I modified the example to output altitude directly from the barometer topic (`TOPIC_ALTITUDE_BAROMETER`) and from the fused GPS topic (`TOPIC_GPS_FUSED`).

While indoors, I'm seeing altitudes of around -60 to -70 from any of the above sources.

While outdoors, I haven't tried OSDK 3.4 yet, but I have used OSDK version 3.3.  While flying several feet above the ground, from `TOPIC_GPS_FUSED` in OSDK v3.3 I get altitudes of around -30.

1) Am I interpreting the numbers incorrectly?  Is there some calculation that needs to be done on the value to get a valid altitude?

2) In what units is the altitude value?  Is it in meters?

jswoboda, 2017-11-30:
I've seen this as well.

amenonDJI, 2017-11-30:
@rpici 
1.) You're reading WGS84 Altitude. 
2.) Unit is in meters.

rpici, 2017-11-30:
@amenonDJI 
What about altitude retrieved from `TOPIC_ALTITUDE_FUSIONED`?  This doesn't always return WGS84 height, does it?  The "FUSIONED" in the name suggests that it could return a WGS84 height, or it could return a height from other sources, e.g. barometer.  Can you clarify this?

eeroniemi, 2017-12-01:
See my comment here, could help someone: https://github.com/dji-sdk/Onboard-SDK-ROS/issues/155#issuecomment-348539107

amenonDJI, 2018-01-24:
@rpici @jswoboda Please reference [this](https://github.com/dji-sdk/Onboard-SDK/wiki/Altitude-Reference-for-different-topics) wiki page for more information regarding Altitude reference

closed 2018-01-24 by amenonDJI