dji-sdk/Onboard-SDK #217: Can't connect drone M210 RTK to DJI assistant 2

ViasVegas, 2017-11-30:
I am trying to connect my M210 RTK to DJI assistant 2, but when i connect my drone to the computer, nothing happen. I have switched on the drone and put the button on the good configuration to connect to a computer by USB cable. I have also made both RTK and M210 firmware update using the remote controller and the DJI GO 4 App. 

So here is my problem : How to do to make the M210 RTK available in Dji assistant 2 so i can enable the API.
--

amenonDJI, 2017-11-30:
@ViasVegas Could you confirm that the M210 switch is in the left most position?

ViasVegas, 2017-12-01:
Yes it is in the left most position, the aircraft is on and linked to the pc with the usbA to usbA cable provided by DJI. But my computer does nt recognize the aircraft and when i launch DJI assistant 2 I dont see my aircraft there neither.

amenonDJI, 2017-12-04:
@ViasVegas Could you let me know the FW version of the M210 RTK and DJI Assistant2?

ViasVegas, 2017-12-04:
The dji assistant 2 is in 1.1.8 and for the aircraft i have no way to know because i can't connect it on Dji assistant and on a mobile device it just says it is on last version.

ViasVegas, 2017-12-04:
And by the way if i switch the button in the middle position, the RTK is detected by dji assistant but i still can't make my drone recognized.

closed 2019-05-28 by hummelslei