dji-sdk/Onboard-SDK #218: M600 TOPIC_HEIGHT_FUSION returns zero when actually flying

wgebers, 2017-11-30:
Using the onboard linux example from the latest API as a base, we are able to get latitude, longitude and altitude.

The altitude (as mentioned in  #216) is coming back as a negative number. 

The height, using TOPIC_HEIGHT_FUSION works correctly using the simulator, but when actually flying the drone, only gives a zero value. 

Which topic should we be using for altitude & / or height?

justwillim, 2017-12-07:
Please use TOPIC_ALTITUDE_FUSIONED instead, HEIGHT_FUSION need relative distance sensor like Vision or ultrasonic

rpici, 2017-12-07:
@justwillim I'm [not clear](https://github.com/dji-sdk/Onboard-SDK/issues/216#issuecomment-348333159) on what the number returned by TOPIC_ALTITUDE_FUSIONED means.

Does it mean:
- meters above the WGS84 reference ellipsoid?
- meters above mean sea level (MSL)?  
- meters above ground level (AGL)?
- something else?

wgebers, 2017-12-07:
@rpici TOPIC_ALTITUDE_FUSIONED as I understand it, is a hybrid of the barometic height and the gps height.  From testing, it appears to be relative to WGS84 and in meters.  There does not appear to be a topic that provides height above ground level / takeoff height, which was my original question.

amenonDJI, 2018-01-24:
@wgebers  Please reference [this](https://github.com/dji-sdk/Onboard-SDK/wiki/Altitude-Reference-for-different-topics) wiki page on more information regarding Altitude reference

tompe17, 2018-04-29:
The ROS SDK uses TOPIC_HEIGHT_FUSION and that works on the M100. But on the M600 we get 0.0 when flying (works in the simulator) on the height_above_takeoff topic.

So is this a bug on the M600 then or is it intended to be this way.

What I need to compute is the height above takeoff for the current position. How would I do that for the M600? I need that to upload a mission waypoint for the current position.

wgebers, 2018-04-29:
@tompe17 I ended up capturing the height at takeoff as a local variable and then computing the height using TOPIC_ALTITUDE_FUSIONED.  As suggested above, the TOPIC_HEIGHT_FUSION requires you to have an external sensor attached to the drone.

tompe17, 2018-04-30:
@wgebers I will do something similar using the ROS API.  The big disadvantage with this is that the programs cannot be restarted in the air without saving some state information on file.  We use a NUC to control the vehicles and it would be good to be able to reboot them in the air and have every state information you need in the controlller.

I do not understand how it can work on the M100 then since the M100 uses TOPIC_HEIGHT_FUSION and do not have any external sensors.

Looking more carefully on the ROS API code I see that for the M600 it seems to be the height field in the global position that is used for height_above_takeoff. For the M100 it is a subscription to the topic TOPIC_HEIGHT_FUSION that is used.  Maybe that is why I see zero on the M600 but not on the M100.

I also see in the ROS API code that they save the reference altitude at takeoff and uses this to compute the local_position. So for me the z value in local_position ought to be the height above takeoff.

closed 2019-05-28 by hummelslei