dji-sdk/Onboard-SDK #219: Waypoint mission status update doesn't work in coordinated trace mode.

eeroniemi, 2017-12-01:
Onboard-SDK version: 3.4 (GitHub commit ID: 955c1eb5b10010a8d64ae7edd6a0f4076721b909)
Onboard-SDK-ROS version: 3.4 (GitHub commit ID: 126def394573a4d32f3a93904f5f879cf6376e08)

I'm building application that relies on OSDK's `setWaypointEventCallback` to return me information when drone has reached each waypoint. Everything works fine when I upload mission with trace mode set to 0 (TRACE_POINT) but I only get first and last waypoint on mode 1 (TRACE_COORDINATED).

Callback function on ROS node (this can be also easily repeated without ROS):

```
void
DJISDKNode::waypointEventCallback(Vehicle *vehicle, RecvContainer recvFrame, DJI::OSDK::UserData userData)
{
  ROS_INFO("wp event callback, \tincident_type: %d\twaypoint_index: %d\tcurrent_status: %d\n",
         recvFrame.recvData.wayPointReachedData.incident_type,
         recvFrame.recvData.wayPointReachedData.waypoint_index,
         recvFrame.recvData.wayPointReachedData.current_status
  );
}
```

When uploading mission in trace mode, everything works fine and I get event callback for each waypoint:

```
[ INFO] [1512141567.874410307]: uploaded the 1th waypoint

[ INFO] [1512141568.900384715]: uploaded the 2th waypoint

[ INFO] [1512141569.920958493]: uploaded the 3th waypoint

[ INFO] [1512141570.940451196]: uploaded the 4th waypoint

[ INFO] [1512141571.960868693]: uploaded the 5th waypoint

[ INFO] [1512141572.978106234]: uploaded the 6th waypoint

[ INFO] [1512141579.978308890]: waypoint mission initialized and uploaded
[ INFO] [1512141598.600343321]: wp event callback, 	incident_type: 2	waypoint_index: 0	current_status: 4

[ INFO] [1512141598.640406278]: wp event callback, 	incident_type: 2	waypoint_index: 0	current_status: 6

[ INFO] [1512141606.699557909]: wp event callback, 	incident_type: 2	waypoint_index: 1	current_status: 4

[ INFO] [1512141607.399744610]: wp event callback, 	incident_type: 2	waypoint_index: 1	current_status: 6

[ INFO] [1512141616.579842124]: wp event callback, 	incident_type: 2	waypoint_index: 2	current_status: 4

[ INFO] [1512141617.279352906]: wp event callback, 	incident_type: 2	waypoint_index: 2	current_status: 6

[ INFO] [1512141624.559282595]: wp event callback, 	incident_type: 2	waypoint_index: 3	current_status: 4

[ INFO] [1512141625.499402220]: wp event callback, 	incident_type: 2	waypoint_index: 3	current_status: 6

[ INFO] [1512141632.239380568]: wp event callback, 	incident_type: 2	waypoint_index: 4	current_status: 4

[ INFO] [1512141633.379596059]: wp event callback, 	incident_type: 2	waypoint_index: 4	current_status: 6

[ INFO] [1512141642.039365187]: wp event callback, 	incident_type: 2	waypoint_index: 5	current_status: 4

[ INFO] [1512141642.079152636]: wp event callback, 	incident_type: 2	waypoint_index: 5	current_status: 6

[ INFO] [1512141642.098986443]: wp event callback, 	incident_type: 1	waypoint_index: 1	current_status: 0
```

When I run exactly the same mission on coordinated mode, I would expect to get callback on every waypoint but I don't:

```
[ INFO] [1512140973.112466479]: initialized waypoint mission
[ INFO] [1512140974.132421079]: uploaded the 1th waypoint

[ INFO] [1512140975.152434449]: uploaded the 2th waypoint

[ INFO] [1512140976.165654658]: uploaded the 3th waypoint

[ INFO] [1512140977.191950615]: uploaded the 4th waypoint

[ INFO] [1512140978.212258617]: uploaded the 5th waypoint

[ INFO] [1512140979.230124083]: uploaded the 6th waypoint

[ INFO] [1512140986.230356116]: waypoint mission initialized and uploaded
[ INFO] [1512141036.091018262]: wp event callback, 	incident_type: 2	waypoint_index: 0	current_status: 4

[ INFO] [1512141073.750581341]: wp event callback, 	incident_type: 2	waypoint_index: 4	current_status: 4

[ INFO] [1512141073.810406998]: wp event callback, 	incident_type: 1	waypoint_index: 1	current_status: 0
```

If you are unable to confirm this issue with information I have provided please let me know and I can provide you code to reproduce the problem.

rohitsantdji, 2017-12-06:
@eeroniemi this is in fact expected behavior, though it is not well documented - apologies for that.

The waypointEventCallback receives triggers from the aircraft only upon certain events during a mission - as the API reference documents, these events (current_status : 4 and current_status: 6) correspond to waypoint actions as documented [here](https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1ACK_1_1WayPointReachedData.html) (although it seems there's a bug in the doc causing it to appear for the next member in the struct).

For TRACE_COORDINATED missions, you are not able to attach waypoint actions (since curved missions don't go exactly through a waypoint, but instead make a curve that passes near the waypoint), so there are no triggers sent by the flight controller except at the first and last waypoints.

eeroniemi, 2017-12-07:
@rohitsantdji Thank you for your answer. In the application we are building it would be really helpful to get information when aircraft has reached certain waypoint also in coordinated mission, even if it didn't hit the exact waypoint coordinates. I'm not sure if this is something that you could consider implementing - this would definitely make DJI based drones more attaractive to our use cases.

rohitsantdji, 2017-12-13:
Thanks for the feedback. The missions system on the flight controller is quite mature and stable at this time, but over a few months' horizon we do have exciting plans for onboard missions. I'll close this for now, but stay tuned!

closed 2017-12-13 by rohitsantdji