dji-sdk/Onboard-SDK #220: Arm N3 without Compass

iLenoy, 2017-12-04:
The project we are currently working on involves our Drone with an N3 flightcontroler to fly indoors with strong magnetic interferance. We are running into the problem, that our N3 will not let us arm on some places on the ground because the compass is encountering to strong interferances. Is there anyway to Arm the N3 even when the compass is not available so we could still take off. We are not using the compass for out positioning or controll system so it would be easiest to just disable the threshold for arming the drone if the compass is not available. Is this possible?

rohitsantdji, 2017-12-06:
Unfortunately, this is not possible as of now, since even in Attitude mode the drone needs the compass information for stabilizing its attitude. One possible workaround could be to calibrate the compass before each flight. It will be an incorrect calibration due to the interference, but it will let you arm the drone, and you can use your external controller to make sure it behaves correctly.

rohitsantdji, 2018-01-10:
Closing due to no follow up from OP. If there are still lingering questions, please re-open.

closed 2018-01-10 by rohitsantdji