dji-sdk/Onboard-SDK #221: M600 + Beagle Bone Black or Tegra TX1 run samples one time

natelannan-osu, 2017-12-04:
on newest OSDK, newest M600 firmware, djiosdk-* runs once and then on subsequent attempts to run produces:

ERROR/1 @ extractOnePackage, L364: Package does not have a valid DataBuffer

natelannan-osu, 2017-12-04:
when attempting to use commands in example:

STATUS/1 @ getErrorCodeMessage, L626: startPackage
STATUS/1 @ getCMDSetSubscribeMSG, L708: SUBSCRIBER_PACKAGE_ALREADY_EXISTS

rohitsantdji, 2017-12-06:
@natelannan-osu this error occurs when the sample doesn't exit cleanly - are you seeing segmentation faults or any other exit-related failures when you run the samples the first time?

rohitsantdji, 2017-12-07:
@natelannan-osu we actually fixed a bug in the telemetry sample that was causing this post OSDK 3.4. Could you pull the latest commit and try again?

natelannan-osu, 2017-12-08:
Sure thing.  sorry for my slow response.  Answering your earlier question, I did not see any segfaults when running the samples the first time.  It reports everything closed successfully.  I will test the latest commit tomorrow when I get to work.

natelannan-osu, 2017-12-08:
I tried 3.4.1.  Running djiosdk-telemetry-sample now produces a segfault due to an error in initVersion:

root@arm:~/Onboard-SDK/build/bin# ./djiosdk-telemetry-sample UserConfig.txt
Read App ID
User Configuration read successfully.
STATUS/1 @ init, L55: Attempting to open device /dev/ttyO2 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L633: Device Serial No. = 0670126493
STATUS/1 @ parseDroneVersionInfo, L635: Hardware = PM820V3
STATUS/1 @ parseDroneVersionInfo, L636: Firmware = 3.2.41.9
STATUS/1 @ getErrorCodeMessage, L644: initVersion
STATUS/1 @ getCommonErrorCodeMessage, L709: ACK_NO_AUTHORIZATION_ERROR
ERROR/1 @ functionalSetUp, L139: Failed to initialize Version! Please exit.
STATUS/1 @ activate, L1086: version 0x3022909
STATUS/1 @ activate, L1099: Activation successful
Segmentation fault

However, when I run 3.4 the first time I am successful:
root@arm:~/Y81000Logger-dev-bbb/Onboard-SDK/build/bin# ./djiosdk-telemetry-sample UserConfig.txt
Read App ID
User Configuration read successfully.
STATUS/1 @ init, L37: Attempting to open device /dev/ttyO2 with baudrate 230400...
STATUS/1 @ init, L47: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L604: Device Serial No. = 0670126493
STATUS/1 @ parseDroneVersionInfo, L606: Hardware = PM820V3
STATUS/1 @ parseDroneVersionInfo, L607: Firmware = 3.2.41.9
ERROR/1 @ functionalSetUp, L193: Virtual RC not supported!
STATUS/1 @ activate, L1057: version 0x3022909
STATUS/1 @ activate, L1070: Activation successful
STATUS/1 @ startPackage, L314: Start package 0 result: 0.
STATUS/1 @ startPackage, L316: Package 0 info: freq=50, nTopics=1.
STATUS/1 @ removePackage, L423: Remove package 0 successful.
STATUS/1 @ initGimbal, L778: Gimbal not mounted!
STATUS/1 @ startPackage, L314: Start package 0 result: 0.
STATUS/1 @ startPackage, L316: Package 0 info: freq=1, nTopics=1.
STATUS/1 @ startPackage, L314: Start package 1 result: 0.
STATUS/1 @ startPackage, L316: Package 1 info: freq=10, nTopics=2.
STATUS/1 @ startPackage, L314: Start package 2 result: 0.
STATUS/1 @ startPackage, L316: Package 2 info: freq=50, nTopics=2.
STATUS/1 @ startPackage, L314: Start package 3 result: 0.
STATUS/1 @ startPackage, L316: Package 3 info: freq=200, nTopics=1.
STATUS/1 @ startPackage, L314: Start package 4 result: 0.
STATUS/1 @ startPackage, L316: Package 4 info: freq=5, nTopics=5.

running a second time is bad news:
root@arm:~/Y81000Logger-dev-bbb/Onboard-SDK/build/bin# ./djiosdk-telemetry-sample UserConfig.txt
Read App ID
User Configuration read successfully.
STATUS/1 @ init, L37: Attempting to open device /dev/ttyO2 with baudrate 230400...
STATUS/1 @ init, L47: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L604: Device Serial No. = 0670126493
STATUS/1 @ parseDroneVersionInfo, L606: Hardware = PM820V3
STATUS/1 @ parseDroneVersionInfo, L607: Firmware = 3.2.41.9
ERROR/1 @ functionalSetUp, L193: Virtual RC not supported!
STATUS/1 @ activate, L1057: version 0x3022909
ERROR/1 @ extractOnePackage, L364: Package does not have a valid DataBuffer
ERROR/1 @ extractOnePackage, L364: Package does not have a valid DataBuffer
ERROR/1 @ extractOnePackage, L364: Package does not have a valid DataBuffer

rohitsantdji, 2017-12-08:
Is this repeatable on 3.4.1 on your system? I see you've received an ACK_NO_AUTHORIZATION_ERROR on connection - what happens when you run it subsequently?

commit d7650f36 (2017-12-14): OSDK 3.4.2: Fix for version segfault. Closes #221.
closed 2017-12-14 by rohitsantdji