dji-sdk/Onboard-SDK #223: waypoint damping

greg-draganfly, 2017-12-15:
Can someone please explain the usage of WaypointSettings.damping?  The comment in the source reads:

float32_t damping; /*!< Bend length (effective coordinated turn mode only) *

What are the units on this?  I want to fly a mission where each waypoint is NOT stopped at, but I don't really care about the smoothness of the curve.  I have had some cases where waypoints return 0xE7.  Can someone explain what this error means and how the waypoints must be configured to avoid it?

jarednm, 2019-04-10:
Did you ever find out?  Wondering the same thing.

lanyusea, 2019-04-10:
the coordinated turn mode means the drone will not fly from point to point directly like polyline but do curve when passing a waypoint.

the bend length is a param to determine how the curve looks like.

jarednm, 2019-04-10:
Sure, but what are the units on damping?  Is it dimensionless 0-1, 0-100%, or is it meters of the radius of curvature, etc?

lanyusea, 2019-04-11:
Forget about the details since I have been away from SDK project for years. I suggest to try it in the simulator and see how it influences the curve.

closed 2019-05-29 by hummelslei