dji-sdk/Onboard-SDK #224: 3.5version  can't  make.    I need help!!!

XMUlqr, 2018-01-03:
root@lqr:/home/lqr/Onboard-SDK# mkdir build
root@lqr:/home/lqr/Onboard-SDK# cd build
root@lqr:/home/lqr/Onboard-SDK/build# cmake ..
-- The C compiler identification is GNU 7.2.0
-- The CXX compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lqr/Onboard-SDK/build
root@lqr:/home/lqr/Onboard-SDK/build# make
Scanning dependencies of target djiosdk-core
[  1%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_ack.cpp.o
[  2%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_broadcast.cpp.o
[  4%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_camera.cpp.o
[  5%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_command.cpp.o
[  7%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_control.cpp.o
[  8%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_error.cpp.o
[ 10%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_gimbal.cpp.o
[ 11%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_hardware_sync.cpp.o
[ 12%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_hotpoint.cpp.o
[ 14%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_mfio.cpp.o
[ 15%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_mission_manager.cpp.o
[ 17%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_mobile_communication.cpp.o
[ 18%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_subscription.cpp.o
[ 20%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_vehicle.cpp.o
[ 21%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_version.cpp.o
[ 22%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_virtual_rc.cpp.o
[ 24%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/api/src/dji_waypoint.cpp.o
[ 25%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/protocol/src/dji_aes.cpp.o
[ 27%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/protocol/src/dji_open_protocol.cpp.o
[ 28%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/protocol/src/dji_protocol_base.cpp.o
[ 30%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_hard_driver.cpp.o
[ 31%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_log.cpp.o
[ 32%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_memory.cpp.o
[ 34%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_platform_manager.cpp.o
[ 35%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/hal/src/dji_thread_manager.cpp.o
[ 37%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/utility/src/dji_circular_buffer.cpp.o
[ 38%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/utility/src/dji_singleton.cpp.o
[ 40%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/platform/default/src/dji_memory_default.cpp.o
[ 41%] Building CXX object osdk-core/CMakeFiles/djiosdk-core.dir/platform/linux/src/linux_serial_device.cpp.o
In file included from /home/lqr/Onboard-SDK/osdk-core/platform/linux/src/linux_serial_device.cpp:33:0:
/home/lqr/Onboard-SDK/osdk-core/platform/linux/inc/linux_serial_device.hpp:97:3: error: ‘fd_set’ does not name a type; did you mean ‘tzset’?
   fd_set m_serial_fd_set;
   ^~~~~~
   tzset
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:197:0,
                 from /usr/include/stdlib.h:279,
                 from /usr/include/c++/7/cstdlib:75,
                 from /usr/include/c++/7/bits/stl_algo.h:59,
                 from /usr/include/c++/7/algorithm:62,
                 from /home/lqr/Onboard-SDK/osdk-core/platform/linux/src/linux_serial_device.cpp:34:
/home/lqr/Onboard-SDK/osdk-core/platform/linux/src/linux_serial_device.cpp: In member function ‘int DJI::OSDK::LinuxSerialDevice::_serialStart(const char*, int)’:
/home/lqr/Onboard-SDK/osdk-core/platform/linux/src/linux_serial_device.cpp:322:14: error: ‘m_serial_fd_set’ was not declared in this scope
     FD_ZERO(&m_serial_fd_set);
              ^
/home/lqr/Onboard-SDK/osdk-core/platform/linux/src/linux_serial_device.cpp:322:14: note: suggested alternative: ‘m_serial_fd’
/home/lqr/Onboard-SDK/osdk-core/platform/linux/src/linux_serial_device.cpp:323:26: error: ‘m_serial_fd_set’ was not declared in this scope
     FD_SET(m_serial_fd, &m_serial_fd_set);
                          ^
/home/lqr/Onboard-SDK/osdk-core/platform/linux/src/linux_serial_device.cpp:323:26: note: suggested alternative: ‘m_serial_fd’
osdk-core/CMakeFiles/djiosdk-core.dir/build.make:734: recipe for target 'osdk-core/CMakeFiles/djiosdk-core.dir/platform/linux/src/linux_serial_device.cpp.o' failed
make[2]: *** [osdk-core/CMakeFiles/djiosdk-core.dir/platform/linux/src/linux_serial_device.cpp.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'osdk-core/CMakeFiles/djiosdk-core.dir/all' failed
make[1]: *** [osdk-core/CMakeFiles/djiosdk-core.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
root@lqr:/home/lqr/Onboard-SDK/build#

zliDJI, 2018-01-03:
Hi, what is your build environment? Ubuntu version?

JeremyMJackson, 2018-02-14:
I am having same issue, any resolution?

kgregorian, 2018-02-23:
add `#include <sys/select.h>` to osdk-core/platform/linux/inc/linux_serial_device.hpp

closed 2019-05-28 by hummelslei