dji-sdk/Onboard-SDK #226: QT 3.5 运行不久就会弹出一个异常框

kezhengtao, 2018-01-10:
[image]
视频已经发送至dev@dji.com

justwillim, 2018-01-10:
Hi kezhengtao，

Github issue only accept English report.

justwillim, 2018-01-10:
@rohitsantdji I think this issue is because the QT sample running under a two thread serial port architecture. But in windows version QSerialPort class already have a multi-thread design. Could you help to fix this issue in the next Version? Since we have this issue for a while

//! @sa https://github.com/dji-sdk/Onboard-SDK/pull/208 https://github.com/dji-sdk/Onboard-SDK/pull/209 https://github.com/dji-sdk/Onboard-SDK/pull/210
these three PR would help.

and @kezhengtao you could use these three PR above try to temporarily fix it.

BR,
William

kezhengtao, 2018-01-10:
I just replace  QSerialPort  with something  in Windows.h . And it's fine now.

rohitsantdji, 2018-01-10:
Could you write about what you replaced it with? That will help the community a lot. Thanks!

kezhengtao, 2018-01-11:
I use win serialport.See the attachment.

closed 2019-05-29 by hummelslei